SpacehuhnTech/esp8266_deauther #852: Multiple libraries were found for "ArduinoJson.h"

mariolpantunes, 2018-05-09:
The projects fail at the link stage when it detects multiples libraries for ArduinoJSON.
I installed ArduinoJSON for other projects, and I believe your code also includes the lib.
Is there an easy way to keep both libraries?

tobozo, 2018-05-09:
Multiple libraries are only warnings so it cannot be the cause of your compilation failure.

mariolpantunes, 2018-05-09:
Where is the complete error code:
Arduino: 1.8.3 (Linux), Board: "Generic ESP8266 Module, Serial, 80 MHz, 80MHz, DIO, 115200, 512K (64K SPIFFS), ck"

Archiving built core (caching) in: /tmp/arduino_cache_220835/core/core_deauther_esp8266_generic_UploadTool_esptool,CpuFrequency_80,FlashFreq_80,FlashMode_dio,UploadSpeed_115200,FlashSize_512K64,ResetMethod_ck_6016a306e69c30335c7e5217ab027e17.a
Multiple libraries were found for "ArduinoJson.h"
 Used: /home/mantunes/Arduino/libraries/ArduinoJson
 Not used: /home/mantunes/.arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/libraries/ArduinoJson
/home/mantunes/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: esp8266_deauther.ino.elf section `.irom0.text' will not fit in region `irom0_0_seg'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Generic ESP8266 Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

tobozo, 2018-05-09:
bad board settings do that, you selected a 512K(64K SPIFFS) and it's too small to fit the deauther.

[check the wiki](https://github.com/spacehuhn/esp8266_deauther/wiki) to find out the best board settings for your ESP

cpapazaf, 2018-05-19:
Got same issue and fixed it by removing the ArduinoJson library I had previously installed through the arduino IDE. How I removed it? I deleted the folder  /home/mantunes/Arduino/libraries/ArduinoJson :)

spacehuhn, 2018-10-15:
https://github.com/spacehuhn/esp8266_deauther/wiki/FAQ#compiler-error-irom0text-will-not-fit-in-region

closed 2018-10-15 by spacehuhn