SpacehuhnTech/esp8266_deauther #853: Network shows, but can't connect

Joshfrog, 2018-05-10:
Pwned network shows, but cannot connect.  Tried to most methods of flashing the chip, even erasing it first.
Possible the password is wrong, or some other issue.

tobozo, 2018-05-10:
you can try the [reset sketch](https://github.com/spacehuhn/esp8266_deauther/tree/master/Reset_Sketch) then reflash the deauther

Joshfrog, 2018-05-10:
Just tried it same issue not sure if its a password issue

Joshfrog, 2018-05-10:
password is deauther right?

tobozo, 2018-05-10:
yep, just remember the first boot is much longer as SPIFFS has to be formatted so give it a minute to be sure ^^

schinfo, 2018-05-10:
Please download and install the esp8266_deauther again.
We found the problem with the AP and Spacehuhn solved it yesterday evening.
Don't forget, to reset the NodeMCU first - https://github.com/spacehuhn/esp8266_deauther/tree/master/Reset_Sketch (Method 3) - that's very important !!!

Joshfrog, 2018-05-10:
Thanks works now cool gadget

closed 2018-05-10 by tobozo