SpacehuhnTech/esp8266_deauther #854: platformio support

bertrik, 2018-05-10:
Hi, perhaps you've heard of platformio, it's a way of compiling an Arduino from the command line, with the advantage that every setting for things like the target platform, included libraries (and their versions) is specified in a project-specific text file instead of having to seek them out in the Arduino IDE menus.

I've seen some other projects even using the platform io build system to build different versions of their firmware, one for each language that they support.

I can probably help create a platformio configuration file.
I just made a very simple one and at least everything compiles.
How about I create a pull request for this?

I don't have all the hardware, like the supported displays, so I cannot test those.

tobozo, 2018-05-10:
There's an [ongoing request](https://github.com/platformio/platform-espressif8266/issues/79) about adding the necessary modified SDK 2.0.0 to the existing platforms.

schinfo, 2018-05-11:
Hello Bertrik,

it would be great, if you can write the configuration file for PlatformIO for the special old SDK 2.0 from tobozo.
My request is from March but nothing was happed since this date.

bertrik, 2018-05-12:
Sorry, I wasn't aware of the need for a patched SDK. I know how to do the simple platformio stuff, like adding a specific version of a lib, but not how to use a different SDK.

bertrik, 2018-05-12:
Below is what I have currently (for a Wemos D1 mini board), with the dependencies fixed at the current most up-to-date versions. This should be saved as platformio.ini in the esp8266_deauther/esp8266_deauther directory (same level as the .ino file). It compiles, but I don't know if it works.

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html

[platformio]
src_dir = .

[env:wemosd1mini]
platform = espressif8266
board = d1_mini
framework = arduino
monitor_baud=115200
lib_deps =
  Adafruit Neopixel@1.1.6
  ArduinoJson@5.13.1
  ESP8266_SSD1306@4.0.0

```

tobozo, 2018-05-12:
Yep it's very strange you don't get any warning or errors at all. Did you patch the arduino environment manually?

If not I'm curious to know if the resulting binary works, what does the serial output say when you flash your D1 Mini?

Here's how to 'downgrade' the SDK to the required version:

- `cd  ~/.platformio/packages`
- `rm -rf framework-arduinoespressif/`
- `git clone https://github.com/tobozo/Arduino framework-arduinoespressif`
- `cd framework-arduinoespressif/`
- `echo 2.0.0 >version.txt`

note that this will affect all further compilations for esp8266, so maybe it's better to put this in a script that does it for you when required

Now on platformio config: ideally there should be a custom package named *framework-arduinoespressif8266deauther* in the *espressif8266* platform.
The doc explains how to do that,

deadgray, 2019-01-18:
Well I moved from ArduinoIDE to PlatformIO finally and found this thread.
Also there is http://docs.platformio.org/en/stable/platforms/creating_platform.html#packages as documentation for start.

deadgray, 2019-01-20:
@tobozo 
Your procedure doesn't seem to work anymore?
Platformio downloads correct package if I git clone your repo into framework-arduinoespressif or framework-arduinoespressif8266. Even when I had 2 of them, symlinked yours and left orginal as blah_org platformio magically finds that _org dir and links to it :-)

tobozo, 2019-03-14:
hey @deadgray not sure what you're expecting, your previous post sounded like you found a fix

closed 2020-09-21 by stale[bot]

omski, 2021-04-14:
I was about to use VS Code with platform.io too. I was able to convert the arduino project, to build and flash the binary to an DStike deauther V6 with VSCode and platform.io but it seems that this build is unable to send death packets. Beacon and Probe is working as expected. The same code built with arduino works without any issues. Any hints what I can do to make it properly build with VS Code and platform.io?