SpacehuhnTech/esp8266_deauther #855: Error compiling for board generic ESP8266 Module

armodrillo250, 2018-05-11:
i followed all the steps and am trying to compile everything but when i do press upload on arduino i get this error message
"Arduino: 1.8.5 (Windows 10), Board: "Generic ESP8266 Module, Serial, 80 MHz, 40MHz, DIO, 115200, 4M (1M SPIFFS), ck"

In file included from sketch\Accesspoints.h:11:0,

                 from sketch\Accesspoints.cpp:1:

sketch\Names.h:10:25: fatal error: ArduinoJson.h: No such file or directory

 #include <ArduinoJson.h>

                         ^

compilation terminated.

exit status 1
Error compiling for board Generic ESP8266 Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
"
I don't really know what this means as i just bought this to mess with some friends could anyone be of some help to me please i would apreciate it thx :D

armodrillo250, 2018-05-11:
i seem to have fixed this issue by installing the arduinojson library but now i am getting this error message.
"Arduino: 1.8.5 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

Build options changed, rebuilding all
In file included from sketch\DigitalLed.h:9:0,

                 from sketch\DigitalLed.cpp:1:

sketch\A_config.h:7:24: fatal error: SH1106Wire.h: No such file or directory

 #include "SH1106Wire.h"

                        ^

compilation terminated.

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
"

tobozo, 2018-05-11:
your environment is incomplete or you selected the wrong board from the arduino menu, make sure the board you selected is part of the "Esp8266 Deauther Modules" list in the Arduino boards menu

armodrillo250, 2018-05-11:
it is supported. i think that i dont have all the required arduino libraries installed i just installed the one containing "SH1106Wire.h" but now i got "SH1106Spi.h" missing.
Could someone supply me with all the required arduino libraries thx

tobozo, 2018-05-11:
yup, as said in the error message:

```
This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

also make sure you haven't messed with the said preferences

spacehuhn, 2018-06-28:
@tobozo made the esp8266 deauther sdk that you can install and get all required libraries with it. You just have to follow the instructions on the wiki and make sure you select one of the deauther boards.

You can of course also rebuild that environment yourself, as we have explained in the repository: https://github.com/tobozo/Arduino/blob/deauther/README.md

closed 2018-10-15 by spacehuhn