SpacehuhnTech/esp8266_deauther #856: In the Boards I cansearch "esp8266" andarduino-esp8266-deauther

hhz1468255512, 2018-05-11:
> Please search for existing (open and closed) issues first to avoid duplicates.  
Also have a look at the [Wiki](https://github.com/spacehuhn/esp8266_deauther/wiki).  
Please put error messages and code ```inside these 3 quotes/grave accents```
(Delete this text when you've read it)

closed 2018-05-11 by schinfo

hhz1468255512, 2018-05-11:
[image]
I've already added the URL，who can help me,please.

tobozo, 2018-05-11:
https://github.com/spacehuhn/esp8266_deauther/wiki/Installation#compiling-using-arduino-ide

hhz1468255512, 2018-05-11:
I am very glad to see the author himself. I was able to install the ESP8266 Deauther Modules board before and succeeded. But these days have been prompted to download errors, and the Deauther  board has disappeared.
 I am in China now. Is it possible that the Internet is restricted?

tobozo, 2018-05-11:
@hhz1468255512 the author is @spacehuhn 

can you access [this file](https://phpsecu.re/esp8266/package_deauther_index.json) ?

the SDK itself is hosted on github, I guess you wouldn't be able to post here if this was blocked

maybe try again ? I sometimes had this issue where I had to try a couple of times by changing the protocol from HTTP to HTTPS or vice-versa

hhz1468255512, 2018-05-11:
Yes，i can 。My browser has turned over the wall。But IDE i have no ideal。 I can't find your board in the Boards Manager .But I found the deauther2.0.0 folder in Arduino15 packages.It's close to two o'clock in the morning, but I still don't know what's wrong.
.

hhz1468255512, 2018-05-11:
I reinstalled the software, and it worked.Thank you all