SpacehuhnTech/esp8266_deauther #857: compile problem

derdude2k, 2018-05-11:
Hi all,

i get this message... im new in Business... ;)

hope someone can help me...

Arduino: 1.8.5 (Windows 10), Board: "Generic ESP8266 Module, Serial, 160 MHz, 80MHz, DIO, 115200, 1M (256K SPIFFS), nodemcu"

Build-Optionen wurden verändert, alles wird neu kompiliert
Archiving built core (caching) in: C:\Users\DIEMAS~1\AppData\Local\Temp\arduino_cache_560054\core\core_esp8266_esp8266_generic_UploadTool_esptool,CpuFrequency_160,FlashFreq_80,FlashMode_dio,UploadSpeed_115200,FlashSize_1M256,ResetMethod_nodemcu_13dc7bf843346c6ddcaa6611bf9f0fb9.a
sketch\Names.cpp.o:(.text._ZNK11ArduinoJson11JsonVariant14variantAsArrayEv[_ZNK11ArduinoJson11JsonVariant14variantAsArrayEv]+0x8): undefined reference to `__cxa_guard_acquire'

sketch\Names.cpp.o:(.text._ZNK11ArduinoJson11JsonVariant14variantAsArrayEv[_ZNK11ArduinoJson11JsonVariant14variantAsArrayEv]+0xc): undefined reference to `__cxa_guard_release'

sketch\Names.cpp.o: In function `ArduinoJson::JsonArray::invalid()':

sketch/Names.cpp:276: undefined reference to `__cxa_guard_acquire'

sketch/Names.cpp:276: undefined reference to `__cxa_guard_release'

sketch\SSIDs.cpp.o: In function `ArduinoJson::JsonArray::invalid()':

C:\Users\die Maschine\Desktop\wifi jammer\esp8266_deauther-master\esp8266_deauther-master\libraries\ArduinoJson\src/ArduinoJson/DynamicJsonBuffer.hpp:67: undefined reference to `__cxa_guard_acquire'

C:\Users\die Maschine\Desktop\wifi jammer\esp8266_deauther-master\esp8266_deauther-master\libraries\ArduinoJson\src/ArduinoJson/DynamicJsonBuffer.hpp:67: undefined reference to `__cxa_guard_release'

sketch\SSIDs.cpp.o:(.text._ZN5SSIDs4loadEv+0x8d): undefined reference to `__cxa_guard_acquire'

sketch\SSIDs.cpp.o:(.text._ZN5SSIDs4loadEv+0x9e): undefined reference to `__cxa_guard_release'

sketch\Settings.cpp.o: In function `ArduinoJson::JsonObject::invalid()':

C:\Users\die Maschine\Desktop\wifi jammer\esp8266_deauther-master\esp8266_deauther-master\libraries\ArduinoJson\src/ArduinoJson/Serialization/JsonSerializerImpl.hpp:56: undefined reference to `__cxa_guard_acquire'

C:\Users\die Maschine\Desktop\wifi jammer\esp8266_deauther-master\esp8266_deauther-master\libraries\ArduinoJson\src/ArduinoJson/Serialization/JsonSerializerImpl.hpp:56: undefined reference to `__cxa_guard_release'

collect2.exe: error: ld returned 1 exit status

exit status 1
Fehler beim Kompilieren für das Board Generic ESP8266 Module.

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.

tobozo, 2018-05-11:
`C:\Users\die Maschine` space in path are known to cause all sorts of problems

`esp8266_deauther-master\libraries\ArduinoJson\src/ArduinoJson/`  mixed forward and backslashes is sign of a mess in the preferences, also this is not the way you install libraries on Arduino

spacehuhn, 2018-06-28:
Any progress on this?

awesomizer18, 2018-10-12:
I have the same issue... any help appreciated greatly.
c:/users/jackson/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Jackson\AppData\Local\Temp\arduino_build_148237/esp8266_deauther.ino.elf section `.irom0.text' will not fit in region `irom0_0_seg'

sketch\Names.cpp.o:(.text._ZNK11ArduinoJson11JsonVariant14variantAsArrayEv[_ZNK11ArduinoJson11JsonVariant14variantAsArrayEv]+0x8): undefined reference to `__cxa_guard_acquire'

sketch\Names.cpp.o:(.text._ZNK11ArduinoJson11JsonVariant14variantAsArrayEv[_ZNK11ArduinoJson11JsonVariant14variantAsArrayEv]+0xc): undefined reference to `__cxa_guard_release'

sketch\Names.cpp.o: In function `ArduinoJson::JsonArray::invalid()':

sketch/Names.cpp:284: undefined reference to `__cxa_guard_acquire'

sketch/Names.cpp:284: undefined reference to `__cxa_guard_release'

sketch\SSIDs.cpp.o: In function `ArduinoJson::JsonArray::invalid()':

C:\Users\Jackson\Documents\Arduino\libraries\ArduinoJson-master\src/ArduinoJson/DynamicJsonBuffer.hpp:67: undefined reference to `__cxa_guard_acquire'

C:\Users\Jackson\Documents\Arduino\libraries\ArduinoJson-master\src/ArduinoJson/DynamicJsonBuffer.hpp:67: undefined reference to `__cxa_guard_release'

sketch\SSIDs.cpp.o:(.text._ZN5SSIDs4loadEv+0x8d): undefined reference to `__cxa_guard_acquire'

sketch\SSIDs.cpp.o:(.text._ZN5SSIDs4loadEv+0x9e): undefined reference to `__cxa_guard_release'

sketch\Settings.cpp.o: In function `ArduinoJson::JsonObject::invalid()':

C:\Users\Jackson\Documents\Arduino\libraries\ArduinoJson-master\src/ArduinoJson/Serialization/JsonSerializerImpl.hpp:56: undefined reference to `__cxa_guard_acquire'

C:\Users\Jackson\Documents\Arduino\libraries\ArduinoJson-master\src/ArduinoJson/Serialization/JsonSerializerImpl.hpp:56: undefined reference to `__cxa_guard_release'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Generic ESP8266 Module.

tobozo, 2018-10-13:
@awesomizer18 can you try to downgrade your ArduinoJSON version to 5.xx instead of 6.xx in the library manager and see if it improves?

awesomizer18, 2018-10-14:
I went through the instructions again and use the simple way with the NodeMCU flasher, works great! I could not get the Arduino to work but that is fine. Thank you for your support!

closed 2018-10-15 by spacehuhn