SpacehuhnTech/esp8266_deauther #858: Unable to find pwned wifi

rogaha, 2018-05-12:
I'm using `Arduino IDE` to flash the `2.0.5` version of the `esp8266_deauther` project. Everything looks normal, however, I'm not able to see the `pwned` when I scan the wifi available. 

Arduino board: `Wireless CH340 NodeMcu V3 Lua ESP8266 WIFI`

```
$ esptool.py -p /dev/cu.SLAB_USBtoUART flash_id
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Uploading stub...
Running stub...
Stub running...
Manufacturer: 68
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
```

```
[======== SYSTEM INFO ========]
RAM usage: 56496 bytes used [69%], 25424 bytes free [31%], 81920 bytes in total

Current WiFi channel: 1
AP MAC address: 80:90:7e:59:63:42
Station MAC address: 00:f1:77:46:a5:c6
SPIFFS: 2761 bytes used [0%], 2946489 bytes free [99%], 2949250 bytes in total
        block size 8192 bytes, page size 256 bytes
Files: 
  /names.json 2 bytes
  /ssids.json 27 bytes
  /autostart.txt 0 bytes
  /settings.json 514 bytes
  /scan.json 24 bytes
[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '1', hidden: false, captive-portal: true
===============================
```

rogaha, 2018-05-12:
[image]

tobozo, 2018-05-12:
Very strange, although the param is `hidden: false` your ESP behaves as if it was.

How many ESP have you flashed with the deauther? 
Does a WiFi scan from another device corroborate this?

As I see you posted a pic of the Arduino IDE, can you eventually try to checkout the latest version of the code from the repo, compile it && upload?

@spacehuhn has recently commited some changes that may indirectly impact this situation.

rogaha, 2018-05-12:
> How many ESP have you flashed with the deauther?

@tobozo this is the first one. 

> Does a WiFi scan from another device corroborate this?

Yes, I've tried from my phone and laptop. 

I also tried using another approach, it didn't work either:

```
$ esptool.py -p /dev/cu.SLAB_USBtoUART write_flash -fm qio 0x0000 ~/Downloads/ESP8266_Deauther_v2.0.5_4MB.bin                                        
esptool.py v2.3.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x004f
Compressed 687232 bytes to 467849...
Wrote 687232 bytes (467849 compressed) at 0x00000000 in 41.4 seconds (effective 133.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

rogaha, 2018-05-12:
Ok, I'm trying with `master` now.

rogaha, 2018-05-12:
@tobozo I'm experiencing the same problem with `master`.

rogaha, 2018-05-12:
is this unexpected? 

```
# scan
Stopped scan
Scan results saved in /scan.json
Removed all APs
Cleared station list
Starting scan for access points (Wi-Fi networks)...
[===== Access Points =====]
AP list is empty :(
Stopped scan
Scan results saved in /scan.json
Stopped scan
Scan results saved in /scan.json
Removed all APs
Cleared station list
Starting scan for access points (Wi-Fi networks)...
[===== Access Points =====]
AP list is empty :(
Stopped scan
Scan results saved in /scan.json
Stopped scan
Scan results saved in /scan.json
Removed all APs
Cleared station list
Starting scan for access points (Wi-Fi networks)...
[===== Access Points =====]
AP list is empty :(
Stopped scan
Scan results saved in /scan.json
Stopped scan
Scan results saved in /scan.json
Removed all APs
Cleared station list
Starting scan for access points (Wi-Fi networks)...
[===== Access Points =====]
AP list is empty :(
Stopped scan
Scan results saved in /scan.json
Stopped scan
Scan results saved in /scan.json
Removed all APs
Cleared station list
Starting scan for access points (Wi-Fi networks)...
# stop
Stopped scan
Scan results saved in /scan.json

```

tobozo, 2018-05-12:
Maybe run the [Reset Sketch](https://github.com/spacehuhn/esp8266_deauther/tree/master/Reset_Sketch) once, sometimes it fixes broken credentials from previous installations.

tobozo, 2018-05-12:
#835 hints at [version 1.5](https://github.com/spacehuhn/esp8266_deauther/releases/tag/v.1.5) working fine when 2.x doesn't

rogaha, 2018-05-12:
still not working... :/

```
SSID: pwned
SSID length: 5
SSID hidden: 0
password: deauther
password length: 8
channel: 1
Scan hidden APs: 1
deauth reson: 1
attack timeout: 300
attack packet rate: 10
client scan time: 15
attack SSID encrypted: 0
use built-in LED: 1
channel hopping: 0
multiple APs: 0
multiple Attacks: 0
mac change interval: 4
1s beacon interval: 0
LED Pin: 2
nameList cleared

Starting WiFi AP:
SSID     : 'pwned'
Password : 'deauther'
-----------------------------------------------

 generating MACs...done

Starting...

```

tobozo, 2018-05-12:
First startup is very slow (formatting SPIFFS, copying webfiles), your test cycle should always be:

1) flash and run the Reset_Sketch
2) flash the sketch you're currently testing
3) open the serial console and wait 1mn

rogaha, 2018-05-12:
Ok, so I came to the conclusion that the `hardware` is defective. I've tried this simple example: 

```
/*
 *  Simple HTTP get webclient test
 */
 
#include <ESP8266WiFi.h>
 
const char* ssid     = "XXXX";
const char* password = "XXXX";
 
const char* host = "wifitest.adafruit.com";
 
void setup() {
  Serial.begin(115200);
  delay(100);
 
  // We start by connecting to a WiFi network
 
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
 
  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  Serial.print("Netmask: ");
  Serial.println(WiFi.subnetMask());
  Serial.print("Gateway: ");
  Serial.println(WiFi.gatewayIP());
}
 
int value = 0;
 
void loop() {
  delay(5000);
  ++value;
 
  Serial.print("connecting to ");
  Serial.println(host);
  
  // Use WiFiClient class to create TCP connections
  WiFiClient client;
  const int httpPort = 80;
  if (!client.connect(host, httpPort)) {
    Serial.println("connection failed");
    return;
  }
  
  // We now create a URI for the request
  String url = "/testwifi/index.html";
  Serial.print("Requesting URL: ");
  Serial.println(url);
  
  // This will send the request to the server
  client.print(String("GET ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" + 
               "Connection: close\r\n\r\n");
  delay(500);
  
  // Read all the lines of the reply from server and print them to Serial
  while(client.available()){
    String line = client.readStringUntil('\r');
    Serial.print(line);
  }
  
  Serial.println();
  Serial.println("closing connection");
}
```

and i didn't work. I'm seeing this log: 
```
Connecting to RGH-Guest
.......................................................................................................................................................................................................................................................................
```

tobozo, 2018-05-12:
Maybe the two antennas are too close to each other and you get interferences?
A faulty unit is also a possibility.

MrAppAndCrap, 2018-05-12:
Hi Rogaha, 
Have you tried using a new Micro USB cable connected to a wall socket? Sometimes the device isn't able to draw enough power with certain cables. I know it sounds a little obvious, but I had the same issue with the SSD 'Pwned' not showing, until I got an up to date Micro USB cable and charger. Once that was done it showed up after a minuet or two.

rogaha, 2018-05-12:
I've changed the `channel` and flashed using the mini `USB` and it's still not working.

rogaha, 2018-05-12:
```
Mounting SPIFFS...OK
Switched to Channel 3
Settings loaded from /settings.json
Settings saved in /settings.json
Device names loaded from /names.json
SSIDs loaded from /ssids.json
Scan results saved in /scan.json
Serial interface enabled
Started AP
[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '3', hidden: false, captive-portal: true
STARTED! \o/
v2.0.5
Executing /autostart.txt
Done executing script

```

rogaha, 2018-05-12:
quick question: not being able to find any `networks` when I execute `scan` from the Serial interface indicates that the `wifi` chip is not working correctly, right? 

is the output below indicating that the `wifi` chip is properly powered ?

```
[======== SYSTEM INFO ========]
RAM usage: 56496 bytes used [69%], 25424 bytes free [31%], 81920 bytes in total

Current WiFi channel: 3
AP MAC address: 00:01:00:00:f0:32
Station MAC address: fc:7c:d9:b3:d8:3a
SPIFFS: 2761 bytes used [0%], 2946489 bytes free [99%], 2949250 bytes in total
        block size 8192 bytes, page size 256 bytes
Files: 
  /names.json 2 bytes
  /ssids.json 27 bytes
  /autostart.txt 0 bytes
  /settings.json 514 bytes
  /scan.json 24 bytes
[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '3', hidden: false, captive-portal: true
===============================
```

rogaha, 2018-05-12:
```
[===== Access Points =====]
AP list is empty :(
Stopped scan
Scan results saved in /scan.json
Stopped scan
Scan results saved in /scan.json
Removed all APs
Cleared station list
Starting scan for access points (Wi-Fi networks)...
# stop
Stopped scan
Scan results saved in /scan.json

```

mgx0, 2018-05-14:
Hi,
same here.

```
# sysinfo
[======== SYSTEM INFO ========]
RAM usage: 60840 bytes used [75%], 21080 bytes free [25%], 81920 bytes in total

Current WiFi channel: 1
AP MAC address: de:4f:22:1c:0a:a4
Station MAC address: b8:2c:ff:3f:68:2c
SPIFFS: 2761 bytes used [1%], 230920 bytes free [98%], 233681 bytes in total
        block size 4096 bytes, page size 256 bytes
Files: 
  /names.json 2 bytes
  /ssids.json 27 bytes
  /autostart.txt 0 bytes
  /settings.json 513 bytes
  /scan.json 24 bytes
[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '1', hidden: false, captive-portal: true
===============================

```

`# scan aps
Stopped scan
Scan results saved in /scan.json
Removed all APs
Cleared station list
Starting scan for access points (Wi-Fi networks)...
[===== Access Points =====]
AP list is empty :(
Stopped scan
Scan results saved in /scan.json
`

plus scan does not seem to find any APs nearby. tested on multiple boards. the 1.6 version works witout a problem

rogaha, 2018-05-14:
My problem seems to be related to the `wifi` chip not being able to power properly. It's interesting that there is no errors on the software layer, the wifi just doesn't work. I tried a simple example, just to connect to my local wifi and it doesn't work either. My board is this one: http://www.instructables.com/id/Getting-Started-With-ESP8266LiLon-NodeMCU-V3Flashi/

rogaha, 2018-05-14:
the board is being powered via a USB wall adapter, it's consuming 3.2v & 65mA.

mgx0, 2018-05-15:
Hi,

the board is this one:
[image]

I'm pretty sure I've tried everything mentioned in all threads, nothing worked

Lumiahna, 2018-05-16:
Hello,
unfortunately, it seems you got a fake board. 
https://raw.githubusercontent.com/wiki/spacehuhn/esp8266_deauther/img/fake_wemos_3.jpg
Please read this page to get the list of supported device: https://github.com/spacehuhn/esp8266_deauther/wiki/Supported-Devices

tobozo, 2018-05-16:
Although this WeM's brand has made bad quality copies of Dstike's ESP+OLED+Buttons builds, the dev kit linked by @mgx0 isn't a copy of any deauther product.

Moreover, owning a bad NodeMCU copycat doesn't imply that *all* versions of the esp8266_deauther won't work on it, and will never justify any proselitism or antagonism for a particular seller.

The esp8266_deauther 1.6 still has very good support for those low quality boards, and provide a low quality deauthing along with a low quality web ui :-)

rogaha, 2018-05-17:
Just a quick update, I've bought another identical `board` and it's working like a charm. I've installed latest code from master (`2.0.5`) and everything worked as expected.

@tobozo congratulations on the work done here! It's incredible how vulnerable we are using these mostly common `wifi & bluetooth` standards. Keep up the great work and thanks for all your support. 

**bottom line**: BUY another board if you experience the issues as me, don't waste your time trying to turn on the `wifi`, since I didn't get any errors I assume that somehow I would need to provide more current to the to the `wifi` chip. I've noticed a delta of `20mA` between the working and not working boards.

closed 2018-05-17 by rogaha

spacehuhn, 2018-05-17:
Nice to see this issue resolved! I would still like to know what causes this issue but I guess we'll never know for sure.

punkplayer1, 2023-04-27:
Had same issue with no original wemos

But fixed with with v2 instead of v3

Set Sketch + Wifi settings

Now works fine :) thanks