SpacehuhnTech/esp8266_deauther #859: Problems with NodeMCU(flashing, communication) after flashing deauther

alekexe, 2018-05-12:
I flashed my NodeMCU 8266 with the latest 1mb bin and after that I could not find pwned SSID or get any communication.

NodeMCU Firmware programmer looks like this after trying again to flash the bin to 8266:
http://alek.online/8266/firmware.png

and ESPlorer looks like this when opening 9600 port:
http://alek.online/8266/esplorer.png

The board itself lights constant red light and it does not make pwned SSID.
What did I do wrong and how can I fix this issuse?
Thanks in advance. :))

alekexe, 2018-05-12:
I have a Windows PC and Linux laptop so if I need to use different software that is not a problem for me. 

I tried flashing through esptool again from laptop and that does not help. I changed 3 USB cables and also I tried it with an Android device and cables seem totally OK..

tobozo, 2018-05-12:
Can you please post a link to the model you're using?

There are simiral issues popping up recently and we're still trying to find out if it's related to a specific NodeMCU model or if it's coming from the code.

As reproductibility on our side is the issue, random tests and observations on your side will be the only significant steps to find a solution.

It does not appear to be system related anyway.

Radom tests include: 
- being able to compile & flash the latest clone of this repo
- having more than one WiFi device to compare scan results with
- fiddling with the settings in the serial console
- running the reset sketch
- looking at similar issues (even closed ones) in the issue tracker of this repo and try to reproduce them

alekexe, 2018-05-12:
Alright, I will try the things that you wrote. Thank you very much, you will be updated ASAP!

alekexe, 2018-05-12:
NodeMCU firmware programmer log:

```
Note:Detect serial port changed.
Note:Auto MAP serial port.Port-->COM6

Note:Serial port connected.
Note:Begin find ESP8266.
```

tobozo, 2018-05-12:
you're flashing an ESP32 with a sketch that was made for an ESP8266 :rofl: 

Closing this as out of scope since you can't use the deauther on an ESP32.

closed 2018-05-12 by tobozo

alekexe, 2018-05-12:
Damn it! :laughing:

tobozo, 2018-05-12:
With your ESP32 you can still try the [Packet Monitor](https://github.com/spacehuhn/PacketMonitor32) though

alekexe, 2018-05-12:
What abot this repo https://github.com/Deflopper/esp32_deauther
This is from where I bought my board, am I just noob or title was misleading?
https://www.kupindo.com/Elektronika/42039117_NodeMCU-ESP8266-CP2102-Wifi-board

tobozo, 2018-05-12:
the product title on this page is just wrong but hey, you got a dual core at 240MHz instead of a monocore at 80MHz :+1: 

The esp32_deauther project compiles fine, but does not deauth anything at all. Some features such as the WiFi scan are working though.

trngkn, 2018-09-21:
> 
> 
> Can you please post a link to the model you're using?
> 
> There are simiral issues popping up recently and we're still trying to find out if it's related to a specific NodeMCU model or if it's coming from the code.
> 
> As reproductibility on our side is the issue, random tests and observations on your side will be the only significant steps to find a solution.
> 
> It does not appear to be system related anyway.
> 
> Random tests include:
> 
>     * being able to compile & flash the latest clone of this repo
> 
>     * having more than one WiFi device to compare scan results with
> 
>     * fiddling with the settings in the serial console
> 
>     * running the reset sketch
> 
>     * looking at similar issues (even closed ones) in the issue tracker of this repo and try to reproduce them
> 
>     * try version 1.5 as hinted in #835

Dear tobozo, i have same issue with alekexe but now i cant connect esp8266 to my laptop. I flashed deauther v2.0.5 by arduino and now my esp8266 not work at all. It not blink led when i connect to pc, not show port on any program to flash it. Wish u can see this and help me, tks u!

p/s: sorry for my bad English, i'm from Vietnam

tobozo, 2018-09-21:
@trngkn the symptoms you describe make it sound like the ESP8266 is dead, do you have another unit you can test with, to make sure it's not caused by a defective USB port or cable ?