SpacehuhnTech/esp8266_deauther #861: ESP8266 goes extraordinarily slow on version 1.6+ of deauther.

ExploiTR, 2018-05-13:
Till version 1.6 my esp8266 was running normally and fast. But, on 2.0 and up it's very and very slow. the web portal takes about 40seconds to open. and then it doesn't work. errors and errors are shown by this.

tobozo, 2018-05-13:
have you tried to open your browser in incognito mode?

also make sure you keep a minimal distance between antennae

ExploiTR, 2018-05-13:
At same physical environment, esp flashed with v1.6 working too fast than the other.

tobozo, 2018-05-13:
do you apply the reset_sketch between installs ? it's recommended to avoid having settings collisions and it also solves a few problems, maybe it'll solve the slow transfer ?

ExploiTR, 2018-05-20:
@tobozo @spacehuhn 
I do the flash using the programmer.

What I did in the day when the problem occurred :1234: 

1. ESP was installed with v1.6, opened it, was working fine
2. Checked the release page found a newer version
3. Flashed the esp with a blank_1MB_bin file
4. Flashed 2.0.5, found it wasn't working, the WebGUI wasn't opening
5. Flashed the esp with a blank_1MB_bin file
6. Flashed 2.1.4 easter, found it wasn't working, the WebGUI wasn't opening, the ping response to 192.168.4.1 was significantly high (about 40-60ms)
7. Flashed the esp with a blank_1MB_bin file
8. Flashed 2.0 \o/, found it wasn't working properly, the GUI took about 40 seconds to open and any tabs weren't working.
9. Flashed the esp with a blank_1MB_bin file
10. Flashed 1.6 and **played some 1/2 an hour with it in peace**

ExploiTR, 2018-05-20:
1. - ESP8266_Deauther_v2.0.5_1MB.bin
    - ESP8266_Deauther_v2.0.0_1MB.bin
    - esp8266_deauther_1mb.bin
2.  _Oops, didn't see the serial output._
3.  **ESP8266 NodeMCU V2** : [Maket Page : Where I bought](https://www.snapdeal.com/product/sunrobotics-nodemcu-v2-lua-based/650520427728#bcrumbSearch:esp8266)
4. Onboard LED

ExploiTR, 2018-05-21:
so @tobozo what's next? Should I give a try again with 2.0.5 by agreeing 
```
Versions 2.x will copy the WebUI to the SPIFFS on first boot, so you may have to wait a couple of minutes before it's fully operational, also it may fail silently and all you can do is flash it again.
This can be troubleshooted using the serial connection.
```
?

tobozo, 2018-05-21:
Maybe try all 2.0.5 versions except the one with the led ring and the deauther monster?

Also check the documentation of your ESP model to see if the internal OLED is using the same pin as the one you see in the source code.

The thing is this ESP model seems to have a general problem with the deauther 2.x and we haven't spotted yet why this happens, so if you're ready to test it in all possible situations, it'll help, but if you'd rather get a NodeMCU v3 and get things working instead, it's fine too.

ExploiTR, 2018-06-05:
Well, somehow setting a `too-high` baud rate (230400) and other settings, I flashed the version 2.0.5.

Serial works well... You obviously were too right, : **Versions 2.x will copy the WebUI to the SPIFFS on first boot, so you may have to wait a couple of minutes before it's fully operational, also it may fail silently and all you can do is flash it again.**

But, the problem is when I'm using the web interface, it's opening finely, but nothing works caused `TimeOutExceptions`

ExploiTR, 2018-06-05:
When I try to scan APS by the serial : 

```
Starting scan for access points (Wi-Fi networks)...
[===== Access Points =====]
ID SSID                             Name             Ch RSSI Enc. Mac               Vendor   Selected

Exception (28):
epc1=0x4021d661 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff1860 end: 3fff1bf0 offset: 01a0

>>>stack>>>
3fff1a00:  00000020 0000003f 3fff1a90 4021d656  
3fff1a10:  3fff3150 3fff07a4 00000000 402074b1  
3fff1a20:  00000000 3fff07a4 00000000 4020740d  
3fff1a30:  3ffe922c 3fff07a4 00000000 00000001  
3fff1a40:  00000000 3fff07a4 3fff1a90 402076c0  
3fff1a50:  3ffe922c 3fff07a4 00000000 402079f9  
3fff1a60:  3fff50a8 0000000f 00000005 3fff3150  
3fff1a70:  0000000f 00000001 3fff31d0 0000000f  
3fff1a80:  00000001 3fff3150 0000000f 00000001  
3fff1a90:  3fff50a8 0000000f 00000000 3fff6200  
3fff1aa0:  0000000f 00000000 00000000 4020740d  
3fff1ab0:  3fff0af8 0000001a 3fff0af8 00000001  
3fff1ac0:  00000001 3fff07a4 00000000 40207b59  
3fff1ad0:  4021eb2d 00000001 3fff07a4 00000001  
3fff1ae0:  3fff07a4 00000001 00000000 40207bc2  
3fff1af0:  00000000 3fff3050 3fff07a4 40207254  
3fff1b00:  3fff3150 000014d7 402070e8 3fff0764  
3fff1b10:  000000ff 00000001 3fff07a4 40214c66  
3fff1b20:  00000000 00000000 3fff0bd0 3fff0af8  
3fff1b30:  3fffdc20 3fff0bbc 40226bf4 3fff0bd0  
3fff1b40:  3ffef9b4 3fff0bbc 00000005 40226c68  
3fff1b50:  00000000 0000000f 00000064 40226cac  
3fff1b60:  402016ae 000003e8 3fff32d0 40226234  
3fff1b70:  3fffdc20 3ffef95c 3fff061c 402193fa  
3fff1b80:  3fffdc20 3ffef95c 3ffef9b4 402252cd  
3fff1b90:  00000000 00000000 3fff085c 402093c1  
3fff1ba0:  3fffdc20 3ffef95c 3fff0bb5 3fff0bbc  
3fff1bb0:  3fffdc20 3ffef95c 3fff0bb5 4021eb08  
3fff1bc0:  40226c15 00000000 00000000 3fff0bbc  
3fff1bd0:  3fffdc20 00000000 3fff0bb5 40226c3d  
3fff1be0:  00000000 00000000 3fff0bd0 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
```

ExploiTR, 2018-06-25:
fixed :trollface:  :angry: :name_badge:

closed 2018-06-25 by ExploiTR