SpacehuhnTech/esp8266_deauther #865: Any lightweight builds (Without OLED,RGB,NEOPIXEL)?

DALVYK, 2018-05-17:
I'm using WEMOS ESP-WROOM-02 board and it works fine. It would be really nice to have a lightweight build (only WEB UI & SERIAL).

spacehuhn, 2018-05-17:
If you don't enable the display in the config, it won't do anything with it. Same with the LEDs. The 512kb, 1mb and 4mb bin files are the "lightweight" builds.

DALVYK, 2018-05-17:
I removed all codes for button,neopixel,digitalled,rgbled,oled. I compiles and runs. Still testing.

spacehuhn, 2018-05-17:
What is your goal with this anyway?

DALVYK, 2018-05-17:
I'm looking if there's any noticeable RAM usage difference and guess there is... only 1% (I facepalmed my self hahaha).

DALVYK, 2018-05-17:
And I also want to add that it works perfectly fine with WEMOS ESP-WROOM-02 w/ batt holder (2MB spi flash memory)

spacehuhn, 2018-06-28:
I'm closing this now. I will improve the way the hardware is en/disabled in the next update. If you have any more questions or feedback on this matter, feel free to comment.

closed 2018-06-28 by spacehuhn