SpacehuhnTech/esp8266_deauther #868: Format

AnthonyFuller, 2018-05-18:
How do I format it?

closed 2018-05-18 by tobozo

AnthonyFuller, 2018-05-18:
Externally? Since I can't access it.

AnthonyFuller, 2018-05-18:
Reopen it.

digmorepaka, 2018-05-18:
You use the serial port. And if you mean externally like its stolen and you don't have access to it, simple answer: You don't.

AnthonyFuller, 2018-05-18:
It isn't stolen. It's just that I messed up flashing it.

tobozo, 2018-05-18:
This command can't be executed from the WebUI if that's what you're asking.
If you can connect via serial, you just send the "format" command and it'll format the SPIFFS.

Otherwise every time you flash the deauther sketch, on next reboot (and only once) the ESP8266 will format the SPIFFS automatically.

AnthonyFuller, 2018-05-18:
nvm about to get putty up and do it through serial

AnthonyFuller, 2018-05-18:
I get this 
[image]

digmorepaka, 2018-05-30:
use esptool to erase it and reflash it