SpacehuhnTech/esp8266_deauther #871: Beacon attack not working properly

sebampuero, 2018-05-20:
**Hardware/Flashing:**
Board: NodeMCU Lolin V3 ESP8266
Using OLED: NO
Using WebServer: YES
Using Serial: NO
Installation mode: bin
Code/binary version: 2.0.5_4mb
Flash size: 4M
Flash method: NodeMCU-flasher
Flash Frequency:	80Mhz
Upload Speed: 230400
Powered by: Powerbank 5V max output 3A

**Description:**
Creating a beacon list of 20 APs and starting the attack works fine. After 10 seconds the attack stops and all the APs disappear. The other attacks (deauth) work fine.

spacehuhn, 2018-06-28:
Have you tested it with different devices? Some show SSIDs differently than others.
If you could post the serial output would also help, when you start the attack and when it stopps after 10 seconds.

spacehuhn, 2018-10-15:
Closing due to no response.

closed 2018-10-15 by spacehuhn