SpacehuhnTech/esp8266_deauther #874: Upload error

Kappie201, 2018-05-21:
When i try to upload using Arduino it gives me a error

Kappie201, 2018-05-21:
Arduino: 1.8.5 (Windows 10), Board: "Arduino Yún"

C:\Users\k\AppData\Local\Temp\Rar$DIa6836.17453\esp8266_deauther\esp8266_deauther.ino:8:30: fatal error: user_interface.h: No such file or directory

   #include "user_interface.h"

                              ^

compilation terminated.

exit status 1
Error compiling for board Arduino Yún.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

this is the error

tobozo, 2018-05-22:
Wrong board and/or incomplete installation, checking the [wiki](https://github.com/spacehuhn/esp8266_deauther/wiki) will help.

Kappie201, 2018-05-22:
Followed every step, i dont know how i fix this.

tobozo, 2018-05-22:
`Arduino: 1.8.5 (Windows 10), Board: "Arduino Yún" `   ...

Kappie201, 2018-05-22:
Is that a fix?

tobozo, 2018-05-22:
one of the recent changes in the documentation says "if this is your first esp8266 project, better try some other less complex project first"

Kappie201, 2018-05-22:
When i use flasher i get a error in the hud. All the json files failed and i cant hack any wifi networks/

tobozo, 2018-05-22:
Yep, and when the problem can be fixed by upskilling the user, we have no other fix than suggesting to get the skills.

Best approach is to show an investigative attitude, because focusing too much on the end result may result in casting shadows on the path to understanding.

What you need here is a better understanding of the Arduino IDE, and some curiosity on the contents of the tools menu, where you'll find choices and begin to ask yourself questions.

[There is a gitter for esp8266](https://gitter.im/esp8266/Arduino) where people can ask basic questions without adding unnecessary noise to an already saturated channel.

Please take the time to get familiar with the Arduino IDE, understand the basic error messages (like the one you're having), try a few basic projects, and then if you find a bug you'll be welcome to open an issue here since it's the reason to exist for this issue tracker.

Also have a look at other issue trackers and see how questions are asked (spoil: not like in a customer service), you'll  understand that people who take the time to reply do it on their free time and don't do it for money :-)

closed 2018-05-22 by tobozo