SpacehuhnTech/esp8266_deauther #875: Error with 2.0.0

domodial, 2018-05-23:
Hello,

Thanks for this projetct.

I have several questions, I have an error number of the type **"ERROR: loading file: run?cmd=save names"** when scanning Aps or Scan Stations. Overall the scan works and deauther also, the result on my AP is good ! He work fine. But the station scan gives nothing.

I compiled with Arduino IDE version ESP8266 Deauther 2.0.0
I have English and German language only.

I see that releases exist like the v2.1.4-easter to try to correct the errors or to have the language of my country. But I can not understand how to inject into Arduino IDE the new features (because I installed Deauther by the board manager).

I imagine that I have to use Source code (zip) which loads the language files, but after applying a Blank_Flash how to go again with a v2.1.4-easter project, I do not arrive because what is sent corresponds to version 2.0.0 of the board manager.

Thanks for your help.

domodial, 2018-05-24:
Please help me, this is Serial Log

[AP] request: /generate_204 NOT FOUND
[AP] request: / OK
[AP] request: / OK
[AP] request: http://192.168.4.1:80/ NOT FOUND
[AP] request: /uma/v2 NOT FOUND
[AP] request: /generate_204 NOT FOUND
[AP] request: /generate_204 NOT FOUND
[AP] request: /style.css OK
[AP] request: /scan.html OK
[AP] request: /web/lang/en.lang OK
[AP] request: /uma/v2 NOT FOUND
[AP] request: /generate_204 NOT FOUND
[AP] request: /service/update2 NOT FOUND
[AP] request: /generate_204 NOT FOUND
[AP] request: /generate_204 NOT FOUND

tobozo, 2018-05-24:
204 errors are caused by Chrome who was fooled into believing your ESP is a real web portal, and it's randomly scanning for known files in order to "guess" what type of access point it's talking to.

From the other OK messages it seems your deauther is working well ^^

domodial, 2018-05-24:
Thanks for reply ;)
How to install 2.1.4 ? because if I install this, I have 2.0.1 in settings UI :(

tobozo, 2018-05-24:
Use the reset sketch between installations, otherwise it will not overwrite with the updated web UI.

domodial, 2018-05-24:
I always do it :(

tobozo, 2018-05-24:
If you flash the 4MB binary, you use the 4MB reset sketch, and so on ?

domodial, 2018-05-24:
Yes, if I flash 4M I reset 4M.

I make a blank-reset with your file.
I download the zip 2.0.4 and I run the esp8266 file to load arduino.
And I send in the chip. But I still have 2.0.1 in the settings.
Also, you have to choose the Generic ESP8266 board under Deauter section.

domodial, 2018-05-24:
Oh no ! I use Reset_Sketch.ino because there is no 4M.ino but just in .bin

tobozo, 2018-05-24:
So much version numbers I'm lost ^_^

The latest binary with most languages is the [v2.0.5](https://github.com/spacehuhn/esp8266_deauther/releases/tag/v2.0.5), and if you compile from the cloned repository you'll even have a few bugfixes (and maybe new bugs too ^^).

Whatever you choose, remember to enable the serial console before flashing so you can see the output of the first boot (can take up to 1mn), and confirm all web files are indeed copied to the SPIFFS folder, along with capturing the full output. 

That will be a better start to troubleshoot version mismatches than randomly poking with versions (especially easter, released on April 1st .. hmmm).

domodial, 2018-05-24:
Thank you very much, I go it !

domodial, 2018-05-24:
Thank you very much for the explanations !! I am now in 2.0.5 and with the language of my country. Wow! In some messages I understood more than a long tutorial.

spacehuhn, 2018-06-27:
Can this be closed?

domodial, 2018-06-27:
yes :)

closed 2018-06-27 by tobozo