SpacehuhnTech/esp8266_deauther #876: Mores Board and SSID/IP

domodial, 2018-05-25:
Hello everyone !

I'm trying to configure 3 boards right now.
Each board is configured in Arduino with a different IP / SSID.

I can connect on the pc / mac / mobile to the APs but when I want to go on the IP 192.168.4.2 or .3 the AP is unreachable. Can you tell me how it works ?

Thank you very much.

domodial, 2018-05-26:
up,

I have this for 3 projects :

**Wifi.h**
DNSServer dnsServer;
IPAddress apIP(192, 168, 4, 1);

DNSServer dnsServer;
IPAddress apIP(192, 168, 4, 2);

DNSServer dnsServer;
IPAddress apIP(192, 168, 4, 3);

**Settings.cpp and Settings.h**
// AP
  setSSID("zog01");

// AP
  setSSID("zog02");

// AP
  setSSID("zog03");

But only Zog01 (192.168.4.1) work :(
Thank you for help.

tobozo, 2018-05-26:
if you customize the IP don't forget to also set the gateway

    IPAddress ip(192, 168, 4, 3);
    IPAddress gateway(192, 168, 4, 3);
    IPAddress subnet(255,255,255,0);
    WiFi.softAPConfig(ip, gateway, subnet);

domodial, 2018-05-26:
Thank you @tobozo !!

domodial, 2018-05-26:
humm...

Like this ?

// Server and other global objects
ESP8266WebServer server(80);
DNSServer dnsServer;
IPAddress apIP(192, 168, 4, 2);
IPAddress gateway(192, 168, 4, 2);
IPAddress netMsk(255, 255, 255, 0);
WiFi.softAPConfig(ip, gateway, subnet);
File fsUploadFile;

domodial, 2018-05-26:
Don't work :(

domodial, 2018-05-26:
Grrr don't work :(

void resumeAP() {
  if (wifiMode != WIFI_MODE_AP) {
    wifiMode = WIFI_MODE_AP;
    wifi_promiscuous_enable(0);
    **WiFi.softAPConfig(ip, gateway, subnet);**
    WiFi.softAP(wifi_config_ssid.c_str(), wifi_config_password.c_str(), wifi_channel, wifi_config_hidden);
    prntln(W_STARTED_AP);
  }

  wifi_config_path = path;
  wifi_config_ssid = ssid;
  wifi_config_password = password;
  setWifiChannel(ch);
  wifi_config_hidden = hidden;
  wifi_config_captivePortal = captivePortal;

  **WiFi.softAPConfig(ip, gateway, subnet);**
  WiFi.softAP(ssid.c_str(), password.c_str(), wifi_channel, hidden);

// Server and other global objects
ESP8266WebServer server(80);
DNSServer dnsServer;
IPAddress **ip**(192, 168, 4, 2);
IPAddress **gateway**(192, 168, 4, 2);
IPAddress **subnet**(255, 255, 255, 0);
File fsUploadFile;

domodial, 2018-05-26:
Oh, All boards work, but the IP for call each board is 192.168.4.1
Then I select board 1, 2 or 3 and it work
where is the logic of single IP?

spacehuhn, 2018-06-28:
Any progress on this?

closed 2018-10-15 by spacehuhn