SpacehuhnTech/esp8266_deauther #878: esp-01 stack error bootloop

Msprg, 2018-05-27:
Hi, i wanted to use the deauther project on esp-01, but when i try to compile it via arduino IDE, it gives me compile error (too small memory, or something similar), so i flashed bin file, but every time it only falls into bootloop, with this in serial:

 ``` ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Mounting SPIFFS...OK
Formatting SPIFFS...OK

Soft WDT reset

ctx: cont 
sp: 3fff1700 end: 3fff1b60 offset: 01b0

>>>stack>>>
3fff18b0:  40001f46 00000000 00000004 ffffffff  
3fff18c0:  34383338 3fff3200 00000008 00000001  
3fff18d0:  00000005 3fff19f0 ffffffff 00000005  
3fff18e0:  00000005 3fff1910 00000000 00000000  
3fff18f0:  00000001 3fff19f0 00000001 3fff32b4  
3fff1900:  00000002 3fff1970 3fff1970 3ffe9651  
3fff1910:  40002514 3fffdd3c 00000000 000020b0  
3fff1920:  3fff19f0 00000002 00000004 00000000  
3fff1930:  00000000 00000000 00000000 3fff18c0  
3fff1940:  00000000 3fff1b90 3fff32b0 40205a33  
3fff1950:  3fff1970 3fff1970 00000004 4010178e  
3fff1960:  000020c0 3fff30a0 00000000 00000000  
3fff1970:  3ffe9650 000020c0 00000000 3fffbff0  
3fff1980:  00000000 00000000 3fff4970 00000000  
3fff1990:  3fff32b0 4010695e 3fff1a60 00000001  
3fff19a0:  3fff19f0 00000001 3fff1b90 402287ee  
3fff19b0:  000020a0 00000001 3fff1a60 40228846  
3fff19c0:  3fff32b0 00000001 3fff1a60 40228c11  
3fff19d0:  3ffe91cc 3fff1a60 00000001 4025bb24  
3fff19e0:  3ffe91cc 3fff1a60 00000001 40228c90  
3fff19f0:  3ffe00ff 3fff1a60 3fff1a10 402279dd  
3fff1a00:  3fff4910 3fff1a60 00000001 4021e326  
3fff1a10:  3ffe9250 00000000 000003e8 40228876  
3fff1a20:  3fff4970 3fff49c0 3fff1a50 40228978  
3fff1a30:  3ffe91cc 00000000 3fff1ac0 4022899c  
3fff1a40:  3fff4900 3fff1ab0 3fff1ac0 4021f3e5  
3fff1a50:  3fff4950 0000000f 0000000e 402287ee  
3fff1a60:  00000000 0000209f 0000209f 40228846  
3fff1a70:  3ffe8758 3fff1ac0 3fff1ac0 40228876  
3fff1a80:  3ffe8758 3fff0754 3fff1ac0 40228978  
3fff1a90:  40101a6d 3fff1ad0 3fff0754 4025bb24  
3fff1aa0:  3ffe8758 3fff0754 3fff0754 4021aba5  
3fff1ab0:  3ffe8758 40232d95 00000000 00000fa0  
3fff1ac0:  3fff4910 0000000f 0000000e 3fff4910  
3fff1ad0:  0000000f 0000000e 3fff5c80 000001df  
3fff1ae0:  000001de 000000ee 000000ef 000000ff  
3fff1af0:  3fff07b8 3fff07d0 3fff0a6c 3fff1b84  
3fff1b00:  00000001 40232c2a 00000001 4025bb24  
3fff1b10:  3fff1b84 3fff0a6c 3fff0754 4021f853  
3fff1b20:  00000000 00000000 00000000 00000000  
3fff1b30:  00000000 00000000 00000000 3fff0b30  
3fff1b40:  3fffdc20 00000000 3fff0b29 40225a02  
3fff1b50:  00000000 00000000 3fff0b40 40100114  
<<<stack<<<
```

Tried ver 2.0.0 and 2.0.5 1MB, also flash with erase bin file, on 2 esp-01 boards.

You can ask what you want, i try to statisfy you.

spacehuhn, 2018-06-27:
Does your ESP-01 maybe only have 512kb? Have you tried that bin file?

Msprg, 2018-06-27:
HI,
Thank you for your response.
By running example sketch checkFlashConfig.ino on the same esp8266, on serial i receive:

```
Flash real id:   00146085
Flash real size: 1048576

Flash ide  size: 1048576
Flash ide speed: 80000000
Flash ide mode:  DIO
Flash Chip configuration ok.
```
wich i think is about 1MB flash. anyway, i tried to flash 512 kb version and,... Surprise, Surprise! A lot more error codes revealed!... actually, take a look on pasteboard: https://pasteboard.co/HrQudJV.png
( i cannot paste it as text anywhere, bacause that chars on the end of that hex string, causing some text glitches) maybe i have defective unit, but i do not think so, because i tried 3 of them, and issues still persist :!

spacehuhn, 2018-06-27:
Very weird, maybe the pin's make problems, happend in the past. Have you tried to edit `A_config.h`?

Maybe disabling all LEDs by turning it to 255 helps.
```
#define LED_PIN_R 255                
#define LED_PIN_G 255   
#define LED_PIN_B 255 
```

Msprg, 2018-06-27:
Yeah, note that I am still not able to compile it trough Arduino IDE, see last part of my first post.

spacehuhn, 2018-06-27:
Can you post the compiler errors you get?

FirstPure, 2018-07-16:
Set,, Tools/ Flash Size 1M (64K SPIFFS)

nikodiakos, 2018-09-14:
Hello,i have the exact same problem.I am trying to compile the project in esp8266(-01) with 1mb memory and the project version 2.0.When i try to flash the precompiled bin heres the output i get from the serial.

/////////////////////////////////////////////////////////////////
Soft WDT reset

ctx: cont 
sp: 3fff1790 end: 3fff1b60 offset: 01b0

>>>stack>>>
3fff1940:  00000000 3fff1b90 3fff32b0 40205970  
3fff1950:  3fff0001 3fff1970 00000004 4010178e  
3fff1960:  00000000 3fff30a0 00000000 00000000  
3fff1970:  3fff19f0 00000001 00000000 3fffbff0  
3fff1980:  000000fb 00000001 3fff4970 00000000  
3fff1990:  3fff32b0 4010695e 3fff1a60 00000001  
3fff19a0:  3fff19f0 00000001 3fff1b90 40202cc9  
3fff19b0:  3fff32b0 00000000 3fff1b90 40202f17  
3fff19c0:  3fff32b0 00000001 3fff1a60 4025bb24  
3fff19d0:  3ffe91cc 3fff1a60 00000001 40225d28  
3fff19e0:  3ffe91cc 00000100 3fff1a10 40227a05  
3fff19f0:  3ffe00ff 3fff1a60 3fff1a10 402279dd  
3fff1a00:  3fff4910 3fff1a60 00000001 4021e31e  
3fff1a10:  3ffe9250 00000000 000003e8 40228876  
3fff1a20:  3fff4970 3fff49c0 3fff1a50 40228978  
3fff1a30:  3ffe91cc 00000000 3fff1ac0 4022899c  
3fff1a40:  3fff4900 3fff1ab0 3fff1ac0 4021f3e5  
3fff1a50:  3fff4950 0000000f 0000000e 402287ee  
3fff1a60:  00000000 0000209f 0000209f 40228846  
3fff1a70:  3ffe8758 3fff1ac0 3fff1ac0 40228876  
3fff1a80:  3ffe8758 3fff0754 3fff1ac0 40228978  
3fff1a90:  40101a6d 3fff1ad0 3fff0754 4025bb24  
3fff1aa0:  3ffe8758 3fff0754 3fff0754 4021aba5  
3fff1ab0:  3ffe8758 40232d95 00000000 00000fa0  
3fff1ac0:  3fff4910 0000000f 0000000e 3fff4910  
3fff1ad0:  0000000f 0000000e 3fff5c80 000001df  
3fff1ae0:  000001de 000000e6 000000e7 000000ff  
3fff1af0:  3fff07b8 3fff07d0 3fff0a6c 3fff1b84  
3fff1b00:  00000001 40232c2a 00000001 4025bb24  
3fff1b10:  3fff1b84 3fff0a6c 3fff0754 4021f853  
3fff1b20:  00000000 00000000 00000000 00000000  
3fff1b30:  00000000 00000000 00000000 3fff0b30  
3fff1b40:  3fffdc20 00000000 3fff0b29 40225a02  
3fff1b50:  00000000 00000000 3fff0b40 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Mounting SPIFFS...OK
Formatting SPIFFS...OK
//////////////////////////////////////////////////////////

When i try your suggestion to set led pins to 255 as spacehun suggested heres the compile error i get.

//////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
Arduino: 1.8.7 (Windows 10), Board: "Generic ESP8266 Module, Serial, 80 MHz, 80MHz, DIO, 115200, 1M (64K SPIFFS), ck"

sketch\DisplayUI.cpp: In member function 'void DisplayUI::setupDisplay()':

DisplayUI.cpp:11:38: error: invalid conversion from 'const char*' to 'const uint8_t* {aka const unsigned char*}' [-fpermissive]

   display.setFont(DejaVu_Sans_Mono_12);

                                      ^

In file included from C:\Users\NikoDiakos\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\libraries\esp8266-oled-ssd1306-4.0.0\src/SH1106Wire.h:34:0,

                 from sketch\A_config.h:7,

                 from sketch\Settings.h:8,

                 from sketch\DisplayUI.h:10,

                 from sketch\DisplayUI.cpp:1:

C:\Users\NikoDiakos\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\libraries\esp8266-oled-ssd1306-4.0.0\src/OLEDDisplay.h:201:10: error:   initializing argument 1 of 'void OLEDDisplay::setFont(const uint8_t*)' [-fpermissive]

     void setFont(const uint8_t *fontData);

          ^

Multiple libraries were found for "ArduinoJson.h"
 Used: C:\Users\NikoDiakos\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\libraries\ArduinoJson
 Not used: C:\Users\NikoDiakos\Documents\Arduino\libraries\ArduinoJson-master
exit status 1
invalid conversion from 'const char*' to 'const uint8_t* {aka const unsigned char*}' [-fpermissive]

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
////////////////////////////////////////////////////////////////////////////////
So i am stuck i have to mention i had no problems with previous release 1.6 and it flashed and functioned sucessfully so i imagine it has to do with the extra features that were added in 2.0

nikodiakos, 2018-09-14:
I now face another problem after changing 
const char DejaVu_Sans_Mono_12[] PROGMEM = {
to
const uint8_t DejaVu_Sans_Mono_12[] PROGMEM = {
as suggested to fix the compilation issue i flashed the firmware sucessfully but now i cannot acess the web interface the ssid shows up fine but the interface doesnt seem to work heres a sample of the serial output
//////////////////////////////////////////////////////////
[AP] request: /index.html OK
[AP] request: /generate_204 NOT FOUND
[AP] request: /gen_204 NOT FOUND
[AP] request: /generate_204 NOT FOUND
[AP] request: /gen_204 NOT FOUND
true/settings.json
[AP] request: /generate_204 NOT FOUND
[AP] request: /gen_204 NOT FOUND
true/settings.json
[AP] request: /index.html OK
[AP] request: /generate_204 NOT FOUND
[AP] request: /gen_204 NOT FOUND
true/settings.json
//////////////////////////////////

i requested the index.html from my phone but it was never served ,other errors appear periodically like request favicon.ico not found.

spacehuhn, 2018-10-15:
This issue became a collection of different problems from different people, cause by different things.

Therefor I recommend reading and following the [Installation Tutorial](https://github.com/spacehuhn/esp8266_deauther/wiki/Installation) carefully (again).
Also have a look at the [FAQ](https://github.com/spacehuhn/esp8266_deauther/wiki/FAQ)

There are also a bunch of [issues](https://github.com/spacehuhn/esp8266_deauther/issues) already related to most of the things mentioned.

closed 2018-10-15 by spacehuhn