SpacehuhnTech/esp8266_deauther #879: i got this error trying to updload from arduino

saspiero, 2018-05-29:
Build options changed, rebuilding all
Archiving built core (caching) in: C:\Users\saspiero\AppData\Local\Temp\arduino_cache_490308\core\core_deauther_esp8266_esp8285_CpuFrequency_80,FlashMode_dout,UploadSpeed_115200,FlashSize_1M512_d0c084e0f7e25556548e798efff46f78.a
c:/users/saspiero/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\saspiero\AppData\Local\Temp\arduino_build_711081/esp8266_deauther.ino.elf section `.irom0.text' will not fit in region `irom0_0_seg'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Generic ESP8285 Module.

tobozo, 2018-05-29:
you're getting close, now you need to free some memory by truncating the MAC address list in [oui.h](https://github.com/spacehuhn/esp8266_deauther/blob/master/esp8266_deauther/oui.h)

saspiero, 2018-05-29:
thank you it worked

closed 2018-06-28 by spacehuhn

donotoverdriveit, 2019-07-22:
Unfortunately, I did not manage it. Where do I have to put the file down?