SpacehuhnTech/esp8266_deauther #880: NodeMCU 12E not available in boards menu.

Dc5000, 2018-05-30:
Hi,
I've added the URL of death boards from this link 
http://phpsecu.re/esp8266/package_deauther_index.json
But I am not able to see NodeMCU 12E board in the boards menu (ESP8266 Death Boards), only NodeMCU 12.

Please help....

tobozo, 2018-05-30:
Make sure you look under the "ESP8266 Deauther Modules" boards group, it's there :-)

[image]

Phoenix1747, 2018-05-30:
You can also try adding `http://arduino.esp8266.com/stable/package_esp8266com_index.json` to your URLs. Generally, you'll find it under "NodeMCU 1.0 (ESP-12E Module)".

Dc5000, 2018-05-31:
Will check and report.

Dc5000, 2018-05-31:
Not Available, reinstalled the boards menu and tried adding them again.

[image]

tobozo, 2018-05-31:
I could reproduce this on Windows, my Linux installation probably needs an update.

The ESP-12E was in the list but the initial entry was [removed](https://github.com/tobozo/Arduino/commit/277b8ea208b692dc35c37727cf58d25951befad5#diff-fce09ad94aad2ba67472cd9142f00146L393) ....

[image]

.. or more exactly [merged](https://github.com/tobozo/Arduino/commit/277b8ea208b692dc35c37727cf58d25951befad5#diff-fce09ad94aad2ba67472cd9142f00146L71) with the ESP12

[image]

So in theory, when you select ESP12, you also get the settings for the ESP-12E, you only have to adjust the flash size.
In practice, I had to inspect all sub options of the menu and select them manually before first compilation (arduino bug on first run?).

Phoenix1747, 2018-05-31:
Try looking under the category `ESP8266 Modules`, not under `ESP8266 Deauther Modules`.

tobozo, 2018-05-31:
@Phoenix1747 have you tried to compile the esp8266_deauther using the method you're suggesting? 

Selecting a board from the "ESP8266 Modules" will certainly result in a compilation error since the native SDK doesn't expose the necessary WiFi functions.

Phoenix1747, 2018-05-31:
@tobozo Oh, I don't think so 
[image]

tobozo, 2018-05-31:
oh good, we can just drop the modified SDK then 🤦‍♂️

Dc5000, 2018-05-31:
@Phoenix1747 
It is available in ESP8266 Modules but not in deauther one.

@tobozo
Does it mean that by choosing NodeMCU 12 from Modules, will it still work on my 12E(are 12 and 12E same now)? If yes, can you please tell me the settings I need to change to upload to 12E?

Phoenix1747, 2018-05-31:
@Dc5000 Can you please try compiling for the normal NodeMCU anyways (not the deauther one); it worked for me.

tobozo, 2018-05-31:
@Dc5000 only you can tell if it works on your NodeMCU, as I said:

>So in theory, when you select ESP12, you also get the settings for the ESP-12E, **you only have to adjust the flash size**.

There are only four different values for flash size, if you don't know: try them all.

Also @Phoenix1747 should realize he's using a manually customized SDK, which you aren't.

Dc5000, 2018-05-31:
@Phoenix1747 It shows an compilation error while compiling.

@tobozo Can you please tell how can I also add it in my IDE?

tobozo, 2018-05-31:
You already added it, just select the ESP12 board and follow the steps in the wiki.

@Phoenix1747 you are right, in some situations it *does* compile, however without the SDK2.0.0 deauthing just won't work.
Since this is a very misleading situation, I'm modifying the SDK 2.0.0 and will soon modify the codebase to make sure people don't run too far in the wrong direction.

Dc5000, 2018-05-31:
@tobozo will it work if I select ESP-12 in deauther modules menu?

FirstPure, 2018-07-16:
SDK

spacehuhn, 2018-10-15:
The ESP-12 is called ESP-12 and not ESP-12 because there are a lot of different modules like ESP-12E, ESP-12F, ESP12S ... and so on.

You can always trial and error! And there is already a good info about the upload setting here: https://github.com/spacehuhn/esp8266_deauther/wiki/Installation#upload-settings

closed 2018-10-15 by spacehuhn

tobozo, 2018-10-16:
The SDK has been updated in order to split Nodemcu v1.0/v0.9 in the boards menu.

You need to manually remove/install the deauther SDK from the Arduino boards manager and restart the IDE in order to see the changes.