SpacehuhnTech/esp8266_deauther #885: No Response from DeAuther 1.6

ExploiTR, 2018-06-04:
As my ESP8266 Nodemcu v2 not supporting the latest releases #861 - I'm still using the v1.6

But, now a new problem appeared there. When I'm trying to send something like `sysinfo` through the `serial monitor` (using custom SDK by @tobozo ) it's returning none of a thing.

When done a button_reset, it shows this and stops

```
SSID: pwned
SSID length: 5
SSID hidden: 0
password: deauther
password length: 8
channel: 1
Default MAC AP: a2:20:a6:1b:a4:89
Saved MAC AP: a2:20:a6:1b:a4:89
MAC AP random: 0
Scan hidden APs: 1
deauth reson: 1
attack timeout: 300
attack packet rate: 10
client scan time: 15
use built-in LED: 1
channel hopping: 0
multiple APs: 0
multiple Attacks: 0
mac change interval: 4
1s beacon interval: 0
LED Pin: 2
Trying to sync the MAC addr with settings
Synced with saved mac addr : a2:20:a6:1b:a4:89
nameList cleared
Attack LED : OFF

Starting WiFi AP:
SSID     : 'pwned'
Password : 'deauther'
-----------------------------------------------

 generating MACs...done

Starting...
```

After that, when a new client connects to `pwned` ,
```

sending AP scan result JSON done
sending AP scan result JSON done
getting settings json (28)

done
getting attacks JSON (239)...done
getting settings json
```

But, no response for those commands sent :-1:

tobozo, 2018-06-04:
Are you powering the ESP from an Android? If so it's no surprise you get funny results :p

ExploiTR, 2018-06-05:
Nope, powering through Windows. I was thinking to use it through Android.

I was using the Arduino serial monitor from windows..

ExploiTR, 2018-06-05:
But, as usual, it's printing out messages to serial about actions done at the web interface, only serial isn't working

```
starting AP scan...
MAC - Ch - RSSI - Encrypt. - SSID - Hidden
0 - b4:75:0e:b2:f4:3d - 6 - -36 - WPA* - KillingTime - 0
scan done
sending AP scan result JSON .done
```

ExploiTR, 2018-06-05:
**SOLVED**

Issue solved after upgrading to version 2.0.5

closed 2018-06-05 by ExploiTR

ExploiTR, 2018-06-06:
@tobozo , haha I & you forgot that the serial interface was added in 2.0, how I would be getting it in 1.6?