SpacehuhnTech/esp8266_deauther #886: Whole 2.0 version bugged

infectedw, 2018-06-06:
Well, i followed whole instruction everything installed properly.

But after trying to scan i set channel on all and get error  ~~~ERROR: timeout loading file run?cmd=scan~~~ but sometimes it find some networks after i select channel, but now moving to "Attack" section, when i press "Deauth" it don't do anything it show me ~~~ERROR: loading file: attack.json.

Sometimes it won't load WiFi hotspot so i have to couple times turn off and then turn on NodeMCU. Please i would like to someone explain maybe i'm doing something wrong. I wanna fix this soon i need it.

The 1.6 version had better UI and didn't had that much errors... Please back old UI.

marek26340, 2018-06-07:
> but now moving to "Attack" section, when i press "Deauth" it don't do anything it show me ERROR: loading file: attack.json.

Yeah, this is quite normal, and I don't know a solution... It happens when you try to attack on a different channel that your ESP is currently (being controlled) at. 

I can't currently even try to explain the rest of your problems, but I did not encounter any of these, except that attack.json.

infectedw, 2018-06-07:
I will make video how it all looks, and post here or somewhere else and set link here. I have tons of errors...

tobozo, 2018-06-07:
[image]

infectedw, 2018-06-07:
First Video: https://vimeo.com/273944772
Second Video: https://vimeo.com/273945836

I hope you guys see enough!

tobozo, 2018-06-07:
Video 1: you just hit the back button and your browser gave you some cache but you're not actually connected to the ESP8266

Video 2: same as video 1 but you seem to lose your temper :p

spacehuhn, 2018-06-07:
[image]

deantonious, 2018-06-08:
Bonvenon al la EIF. Bonvolu RTFM kaj ekzekutu la komandon "rice" dum sur la seria interfaco por pliaj elpurigado.

closed 2018-10-16 by spacehuhn