SpacehuhnTech/esp8266_deauther #887: What's the highest packet rate that esp8266 can send?

WYHSniper, 2018-06-09:
I tried 150 pkt/s and it works, anyone test it a higher rate?

tobozo, 2018-06-09:
see #717

mdemeter27, 2018-06-11:
How do you change the packet rate?

tobozo, 2018-06-11:
@mdemeter27 you can [play with the settings](https://github.com/spacehuhn/esp8266_deauther/wiki/Serial#improve-packet-rate)

spacehuhn, 2018-06-28:
There is no standard answer to this. It's highly dependent on the network traffic around you, what kind of packets you're sending, at what speed, with what size.

For the deauther, you can play around with if but the higher you set it, the more unstable the whole ESP can get! And it will only send what it can, not more. But you can see how much packets it's sending and adjust it by that. 

Also note that sending more packets, does not necessarily make anything 'better'.

closed 2019-10-09 by tobozo