SpacehuhnTech/esp8266_deauther #888: Interval for continous script run is ignored - fix inside

tobybear, 2018-06-10:
The interval for running a script continuously via "run script.txt -c 60s" is ignored and always set to 10 seconds. This is because line 867 in SerialInterface.cpp contains an error:
``` if (list->size() == 3) {```
It should be:
``` if (list->size() > 3) {```

(if a time interval is given, the size of the liste will be 4, not 3)

tobozo, 2018-06-10:
nice catch :clap: 
Maybe `if (list->size() == 4)` would work?
Also this would need to be changed too:  `continueTime = getTime(list->get(4));`

tobybear, 2018-06-10:
@tobozo : 
>Maybe if (list->size() == 4) would work?
That would be less future-proof if we would introduce more parameters for this function. ">3" is in my opinion slightly better.

>Also this would need to be changed too: continueTime = getTime(list->get(4));
No, this one is correct with get(3), as that actually gets the fourth element (index starts at 0)

spacehuhn, 2020-06-16:
For the [upcoming version 3](https://github.com/SpacehuhnTech/esp8266_deauther/tree/v3), I'm using [SimpleCLI](https://github.com/spacehuhn/SimpleCLI) to solve this issue.

closed 2020-06-16 by spacehuhn