SpacehuhnTech/esp8266_deauther #889: espcomm_open_error compile error

moltenteam, 2018-06-11:
Hi,
today I received my ESP8266 Lolin Module and I decided to upload this software to it.

I have some issues, successfully flashed firmware, but the Arduino sketch I can't upload. It still keeps showing me this error messages: error: espcomm_open error, espcomm_sync error. I followed all the instructions closely and I tried to follow the failsafe upload settings. None of them worked. I tried to customize upload settings by following the module creator and NONE OF THEM WORKED!!!!

Thanks

spacehuhn, 2018-06-11:
Then it's a problem unrelated to the settings: https://github.com/spacehuhn/esp8266_deauther/wiki/Troubleshooting#espcom-error

moltenteam, 2018-06-12:
I installed the drivers you recommend and the drivers from the creator of the module. I checked all settings, it displayed this: Arduino: 1.8.5 (Windows 10), Vývojová deska: "Generic ESP8266 Module, Serial, 160 MHz, 80MHz, DIO, 9600, 4M (3M SPIFFS), nodemcu"

Archiving built core (caching) in: C:\Users\ASUS\AppData\Local\Temp\arduino_cache_441962\core\core_deauther_esp8266_generic_UploadTool_esptool,CpuFrequency_160,FlashFreq_80,FlashMode_dio,UploadSpeed_9600,FlashSize_4M3M,ResetMethod_nodemcu_f1e85eb445eeae63f2b853c48f8e801e.a
Projekt zabírá 683250 bytů (65%)  úložného místa pro program. Maximum je 1044464 bytů.
Globální proměnné zabírají 42216 bytů (51%)  dynamické paměti, 39704 bytů zůstává pro lokální proměnné. Maximum je 81920 bytů.
Uploading 687392 bytes from C:\Users\ASUS\AppData\Local\Temp\arduino_build_896097/esp8266_deauther.ino.bin to flash at 0x00000000
error: failed sending 1070 bytes
.error: failed sending 1044 bytes
.error: failed sending 1040 bytes
.error: failed sending 1040 bytes
.error: failed sending 1083 bytes
.error: failed sending 1064 bytes
.error: failed sending 1079 bytes
.error: failed sending 1059 bytes
.error: failed sending 1062 bytes
.error: failed sending 1069 bytes
.error: failed sending 1058 bytes
.error: failed sending 1054 bytes
.error: failed sending 1053 bytes
.error: failed sending 1061 bytes
.error: failed sending 1066 bytes
.error: failed sending 1064 bytes
.error: failed sending 1069 bytes
.error: failed sending 1054 bytes
.error: failed sending 1057 bytes
.error: failed sending 1051 bytes
.error: failed sending 1051 bytes
.error: failed sending 1057 bytes
.error: failed sending 1057 bytes
.error: failed sending 1061 bytes
.error: failed sending 1072 bytes
.error: failed sending 1059 bytes
.error: failed sending 1059 bytes
.error: failed sending 1054 bytes
.error: failed sending 1052 bytes
.error: failed sending 1047 bytes
.error: failed sending 1044 bytes
warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
warning: espcomm_send_command: didn't receive command response
vybraný seriový port warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
 neexistuje a nebo váš board není připojen

LED onboard is blinking while i upload, btw.

moltenteam, 2018-06-12:
Your failsafe/recommended settings: Arduino: 1.8.5 (Windows 10), Vývojová deska: "Generic ESP8266 Module, Serial, 160 MHz, 80MHz, DIO, 115200, 1M (256K SPIFFS), nodemcu"

Archiving built core (caching) in: C:\Users\ASUS\AppData\Local\Temp\arduino_cache_441962\core\core_deauther_esp8266_generic_UploadTool_esptool,CpuFrequency_160,FlashFreq_80,FlashMode_dio,UploadSpeed_115200,FlashSize_1M256,ResetMethod_nodemcu_f1e85eb445eeae63f2b853c48f8e801e.a
Projekt zabírá 683250 bytů (89%)  úložného místa pro program. Maximum je 761840 bytů.
Globální proměnné zabírají 42216 bytů (51%)  dynamické paměti, 39704 bytů zůstává pro lokální proměnné. Maximum je 81920 bytů.
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_open failed

moltenteam, 2018-06-12:
Sorry, I'm Czech speaker.

tobozo, 2018-06-12:
settings look good, maybe your cable is too long or your USB port is too weak?

the first upload fail probably left a zombie process and letting your COM port busy, you may want to unplug/replug the ESP and select the COM port again, or maybe reboot and retry to make sure this wasn't the case ?

moltenteam, 2018-06-18:
my cable is 50cm long, USB port ? I have USB 3.0 port on.

> the first upload fail probably left a zombie process and letting your COM port busy, you may want to unplug/replug the ESP and select the COM port again, or maybe reboot and retry to make sure this wasn't the case ?

I'll try this.

moltenteam, 2018-06-18:
It's working!!! Thank you!!!

closed 2018-06-18 by moltenteam