SpacehuhnTech/esp8266_deauther #891: when not using USE_PROGMEM_WEB_FILES, wifi.h needs to avoid referncing the consts

bradanlane, 2018-06-14:
This is a minor point which most will not encounter but ...

if you comment out the definition of `USE_PROGMEM_WEB_FILES` from `webfiles.h` the code in wifi.h is not aware the file strings no longer exist. The compiler throws an error.

The fix is not as easy as referencing the definition of `USE_PROGMEM_WEB_FILES` because its not available outside of `esp8266_deauther.ino`.

bradanlane, 2018-06-16:
There also seems to be some issue with page loads once the compressed `gz` files are moved to the spiffs. I frequently get the first page to load in my web browser but a subsequent page will not completely load. It appears the `.html.gz` file is loaded but one of the referenced files (css, js, etc) never completes. Disconnecting from the `ESP8266` AP and reconnecting usually fixes it. 

Have others moved the web interface to spiffs with success or issues?

spacehuhn, 2018-10-17:
I added the missing #ifdef in wifi.h so it completley ignored the progmem files but i would not recomment that. 

You can of course do that and then upload your own files to the SPIFFS with this plugin: https://github.com/esp8266/arduino-esp8266fs-plugin

But it's better to use the converter script and use the PROGMEM, because the SPIFFS isn't as efficiant as the PROGMEM and the website can be a bit unstable.

bradanlane, 2018-10-17:
I eventually made the same conclusion. While I was successful at uploading web files to SPIFFS, it was not as stable - occasional resource load timeouts at the browser.

_The reason for attempting to move the web resources out of PROGMEM was to allow for a few additional resources (images)._

spacehuhn, 2018-10-17:
Yeah I suppose images aren't that great for improving performance either :D

tobozo, 2018-10-17:
just throwing a rock into water: images could be served from spiffs while other assets are still served from PROGMEM?

bradanlane, 2018-10-17:
Good suggestion. Thanks.

Ultimately, I created the graphic I needed using pure CSS.

closed 2019-10-09 by tobozo