SpacehuhnTech/esp8266_deauther #896: issue with trying to launch esp8266_deauther.ino

matchu20002, 2018-06-15:
hi there
I'm trying to launch esp8266_deauther.ino in Ardunion but I keep getting this error message:

```Arduino: 1.8.5 (Mac OS X), Board: "NodeMCU 0.9 (ESP-12 Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/matthewdodd/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/matthewdodd/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/matthewdodd/Documents/Arduino/libraries -fqbn=esp8266:esp8266:nodemcu:CpuFrequency=80,UploadTool=esptool,UploadSpeed=115200,FlashSize=4M3M -ide-version=10805 -build-path /var/folders/5z/mf9v8tsx5bx50vp9z46fyndw0000gn/T/arduino_build_247727 -warnings=none -build-cache /var/folders/5z/mf9v8tsx5bx50vp9z46fyndw0000gn/T/arduino_cache_682383 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=/Users/matthewdodd/Library/Arduino15/packages/esp8266/tools/esptool/0.4.6 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/Users/matthewdodd/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=/Users/matthewdodd/Library/Arduino15/packages/esp8266/tools/mkspiffs/0.1.2 -verbose /Users/matthewdodd/Downloads/esp8266_deauther-master/esp8266_deauther/esp8266_deauther.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/matthewdodd/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/matthewdodd/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/matthewdodd/Documents/Arduino/libraries -fqbn=esp8266:esp8266:nodemcu:CpuFrequency=80,UploadTool=esptool,UploadSpeed=115200,FlashSize=4M3M -ide-version=10805 -build-path /var/folders/5z/mf9v8tsx5bx50vp9z46fyndw0000gn/T/arduino_build_247727 -warnings=none -build-cache /var/folders/5z/mf9v8tsx5bx50vp9z46fyndw0000gn/T/arduino_cache_682383 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=/Users/matthewdodd/Library/Arduino15/packages/esp8266/tools/esptool/0.4.6 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/Users/matthewdodd/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=/Users/matthewdodd/Library/Arduino15/packages/esp8266/tools/mkspiffs/0.1.2 -verbose /Users/matthewdodd/Downloads/esp8266_deauther-master/esp8266_deauther/esp8266_deauther.ino
Using board 'nodemcu' from platform in folder: /Users/matthewdodd/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0
Using core 'esp8266' from platform in folder: /Users/matthewdodd/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0
Detecting libraries used...
"/Users/matthewdodd/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/Users/matthewdodd/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10805 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266 "-I/Users/matthewdodd/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0/cores/esp8266" "-I/Users/matthewdodd/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0/variants/nodemcu" "/var/folders/5z/mf9v8tsx5bx50vp9z46fyndw0000gn/T/arduino_build_247727/sketch/esp8266_deauther.ino.cpp" -o "/dev/null"
"/Users/matthewdodd/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/Users/matthewdodd/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10805 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266 "-I/Users/matthewdodd/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0/cores/esp8266" "-I/Users/matthewdodd/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0/variants/nodemcu" "-I/Users/matthewdodd/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0/libraries/EEPROM" "/var/folders/5z/mf9v8tsx5bx50vp9z46fyndw0000gn/T/arduino_build_247727/sketch/esp8266_deauther.ino.cpp" -o "/dev/null"
"/Users/matthewdodd/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/Users/matthewdodd/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10805 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266 "-I/Users/matthewdodd/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0/cores/esp8266" "-I/Users/matthewdodd/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0/variants/nodemcu" "-I/Users/matthewdodd/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0/libraries/EEPROM" "-I/Users/matthewdodd/Documents/Arduino/libraries/ArduinoJson/src" "/var/folders/5z/mf9v8tsx5bx50vp9z46fyndw0000gn/T/arduino_build_247727/sketch/esp8266_deauther.ino.cpp" -o "/dev/null"
"/Users/matthewdodd/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/Users/matthewdodd/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10805 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266 "-I/Users/matthewdodd/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0/cores/esp8266" "-I/Users/matthewdodd/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0/variants/nodemcu" "-I/Users/matthewdodd/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0/libraries/EEPROM" "-I/Users/matthewdodd/Documents/Arduino/libraries/ArduinoJson/src" "-I/Users/matthewdodd/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0/libraries/Wire" "/var/folders/5z/mf9v8tsx5bx50vp9z46fyndw0000gn/T/arduino_build_247727/sketch/esp8266_deauther.ino.cpp" -o "/dev/null"
"/Users/matthewdodd/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/Users/matthewdodd/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10805 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266 "-I/Users/matthewdodd/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0/cores/esp8266" "-I/Users/matthewdodd/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0/variants/nodemcu" "-I/Users/matthewdodd/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0/libraries/EEPROM" "-I/Users/matthewdodd/Documents/Arduino/libraries/ArduinoJson/src" "-I/Users/matthewdodd/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0/libraries/Wire" "-I/Users/matthewdodd/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0/libraries/SPI" "/var/folders/5z/mf9v8tsx5bx50vp9z46fyndw0000gn/T/arduino_build_247727/sketch/esp8266_deauther.ino.cpp" -o "/dev/null"
"/Users/matthewdodd/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/Users/matthewdodd/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-function [... cut ...]
Using library Wire at version 1.0 in folder: /Users/matthewdodd/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0/libraries/Wire 
Using library SPI at version 1.0 in folder: /Users/matthewdodd/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0/libraries/SPI 
exit status 1
Error compiling for board NodeMCU 0.9 (ESP-12 Module).```

I'm using version 2.0.0  Also I'm using a mac.  I wondering if its to do with the path not being recognised?  any help would be much appreciated!

Thanks

Matthew

tobozo, 2018-06-15:
could it be related to #764 ?

spacehuhn, 2018-06-27:
Have you installed our ESP8266 Deauther SDK in the boards manager and selected the correct board?

Charanmakkar, 2018-06-27:
same ERR0R with me,
i am trying to do wifi deauthering using nodeMCU(1.0 (12E) ) .

please can u help me with correct program .

matchu20002, 2018-06-28:
hi spacehuhn, I'm using Arduion 1.8.5. I'm also using 'ESP8266 2.0.0' in the Boards Manager and the 'Generic ESP8266 Module'.

Is this the correct settings? I'm wondering if its a mac issue as I'm trying to connect using a mac.

spacehuhn, 2018-10-16:
The correct settings are explained here: https://github.com/spacehuhn/esp8266_deauther/wiki/Installation

craftycram, 2018-10-25:
Well I added the correct URLs to the IDE settings, but if i search for arduino-esp8266-deauther there is no library with that name.. :/ 
any clue how I can fix this?

craftycram, 2018-10-25:
okay forget it, I just found my failure. I looked for the aurdino-esp8266-deauther in the library manager. now i found it in the board manager... 🤦‍♂️

fdkjgudogh, 2019-04-13:
why does mine keep saying"oui.h no such file or directory"

closed 2019-10-09 by tobozo