SpacehuhnTech/esp8266_deauther #897: Issue after flashing

NIEIXIUIS, 2018-06-17:
The network does not show up. Ive tried different ways and i have made it connect with first ereasing all the data on the board then running a wifi connect script witch made the nodemcu esp8266 connect to my wifi. Then i uploaded the v1.6 file and it worked nice. Suddenly it stops working and it says no internet access and doesn't let me connect. When i cycled to another place in my city it started working, then i cycled a little back and it stopped working, and when i cycled back again and it worked. It seems like it's position based. But at home it doesn't always work either. When i first upload the code using the way mentioned at the start it works for a few hours and then it stops working, it won't let me connect. If im aleready connected to the esp8266 and it stops working it will me an error during ap scan and then disconnect. I've tried all the other versions with no luck. I just tried flashing the v2.05 version and this is the serial output i got.  

ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

This output was repeated till i unplugged it. Anyone have some suggestions?

cjs1502, 2018-06-27:
i have the same problem.please let me know if you get through this problem.

tobozo, 2018-06-27:
sometimes issuing the "reset" command from the serial console and waiting a few seconds solves it

newlearnergrgsm, 2018-07-01:
hello. ı am sorry for my bad english . ı dont know which issue to write sorry for missunderstand.
ı would like to ask, ı have loaded ESP8266 with flash program and it shows AP MAC and  ST MAC. 
How can I change it. !! 
Thanks for any answer.

spacehuhn, 2018-07-02:
The access point or the web server (probably both) have performance problems that were solved in a later version of the SDK, that we unfortunately can't use for this project. I noticed that there are heavy performance differences depending on how much other WiFi devices or networks are in the area.

Some phones also don't like connecting to a network without internet access and switch to other networks, that doesn't help either...

That's why I want to concentrate more on the display UI, as that doesn't have to deal with such problems.

bradanlane, 2018-07-02:
I would like to "dig into" the web performance and stability issue. Adding a display is not an option in my case. 

Could you outline the key reason the newer version of the SDK cannot be used?

tobozo, 2018-07-02:
@bradanlane The newer SDK does not expose the function needed to send deauth frames and this SDK is closed source.

bradanlane, 2018-07-02:
Thanks for the background.

closed 2018-10-16 by spacehuhn