SpacehuhnTech/esp8266_deauther #900: WiFi Off

cadencetheking, 2018-06-19:
My esp8266 is perfectly functioning and the program works fine. the problem I had is like a dumbass I decided to show one of my friends. In settings he turned off the WiFi when I gave him my phone. I can't not turn it back on because I'm not connected to the network and I can't connect to the network because the WiFi is off. I thought of re uploading in but I don't have enough storage on it. so if someone could tell me how to turn the network back on or if not, tell me how to remove the old program. Thank you for the support

bradanlane, 2018-06-19:
If you can "upload" then it means you can connect the ESP8266 to your computer via USB. You can control the settings via the serial monitor window of the Arduino IDE.

cadencetheking, 2018-06-19:
@bradanlane And what command would I send to enable the network

bradanlane, 2018-06-19:
Start with "help" to see the list of commands as outlined in the wiki:

https://github.com/spacehuhn/esp8266_deauther/blob/master/serialcommands.md

The web interface can be turned on/off as described in the settings document:

https://github.com/spacehuhn/esp8266_deauther/blob/master/settings.md

cadencetheking, 2018-06-19:
@bradanlane thank you for the help

Enziak, 2018-06-20:
Or you can use Serial USB Terminal on phone [https://play.google.com/store/apps/details?id=de.kai_morich.serial_usb_terminal](url)

spacehuhn, 2018-06-27:
Were you able to solve it?

closed 2019-10-09 by tobozo