SpacehuhnTech/esp8266_deauther #901: Serial Interface Messy Outputs :

ExploiTR, 2018-06-24:
Using the serial interface and it just works as fine as it's programmed to do. But, at first start of 8266 I'm getting some messy messages in the console which is quite disturbing. Can you tell where the problem exists? **I'm getting other messages in human-recognisable string as usual**

Like this :
```
��ʼ���pc�&��ak�����p�Q��$�������L���Qr����+����6�>/P=K��U}���g| ����6�?B�`o������ld�W�r8Z��y*^�}W����y��dي���$א���R����嵢�AyH��X�����5
    ����w�~�����F8�k�0��� m������v�P�?>���Y�	�Oy���oz3�3����S@�,�4)��7���stce����"^)�? �9�������¿��?������{q�\�)wR�]����[`��@�B�`��݈��/Z�`��be�
    ���������@��������C�\� �A>���3[�����Ek���C5$�	��n~m�F(��9����b���sN��fYd�����O���A~��d�����W���4�����F���j[����ӡ��k�ٲ����9���)�q�
    �b�|~&VCoL25k�(jN�,_��A�} Aۃ��o������;U��qw�y0���<�Qǈ����Xݤ�C>Lidk���YwR�^���P�֒��mAg�s5��������B|�-����	�
    |g�w!'�1� �C���
```

ExploiTR, 2018-06-24:
both are set to `115200` :crying_cat_face: 
flash size 4M(1M spiffs) [I've verified it with custom sketch]
Freq : 80MHz Mode: DIO Cpu:160

And, everything other is working fine like, `sysinfo` or `scan`

ExploiTR, 2018-06-24:
@bradanlane  and one thing to check , even spacehunn himself is receiving this : -

[image]

spacehuhn, 2018-06-24:
@ExploiTR @ExploiTR @ExploiTR that is normal @ExploiTR

EMLGaming, 2018-06-24:
is it fixed @ExploiTR @ExploiTR @ExploiTR @ExploiTR @ExploiTR ?

ExploiTR, 2018-06-25:
@spacehuhn @spacehuhn  I'm not telling it a bug/feature or anything else. I want to know which line of the program is making this output, will you please tell me? 

If it's not a feature without which it will totally stop working, I'll remove that line and shall then re-compile.  @spacehuhn

jLynx, 2018-06-25:
None, it's part of the chip itself @ExploiTR

ExploiTR, 2018-06-25:
Really? @jLynx  But, when I use a simple sketch it doesn't happen. 

But, in which logic you're saying to set it to `9600` I don't understand. The code itself uses `Serial.begin( 115200 );`

jLynx, 2018-06-25:
Yes but the kernel is 9600. Like I said above. It's not in the code

ExploiTR, 2018-06-25:
@spacehuhn If you can help please do.

closed 2018-06-25 by jLynx

spacehuhn, 2018-06-25:
I can't

ExploiTR, 2018-06-25:
Aaah, I found that the problem is specifically immutable. It's coming from the 2102 (might be). Anyway thanks for the support. It took a large time to make it understand :trollface: 

Can you tell me @spacehuhn @spacehuhn , if I use Arduino with the ESP12 will it also give those messages ???

spacehuhn, 2018-06-27:
As jLynx said:
> None, it's part of the chip itself @ExploiTR so you cannot remove it. It's actually readable if you set the console to 9600 but then you won't be able read the actual messages from the deauther code

When the ESP8266 boots it's printing with 9600 baud. That is part of the bootloader I guess. You can't just disable that. Then it's switchting to 115200 baud because I specified that in the code. There's nothing more to it. It's not a bug.