SpacehuhnTech/esp8266_deauther #904: ESP can't detect stations

ExploiTR, 2018-06-26:
I've 4 devices connected to victim AP. When I perform scan through serial (I've tried all possible commands)

it gives : 
```
Scanning WiFi [20%]:   0 packets/s |  0 devices |  0 deauths
Scanning WiFi [40%]:   0 packets/s |  0 devices |  0 deauths
Scanning WiFi [60%]:   0 packets/s |  0 devices |  0 deauths
Scanning WiFi [80%]:   0 packets/s |  0 devices |  0 deauths
```

Can't understand where's the problem. AP is detected as normal.

tobozo, 2018-06-26:
make sure victim AP uses 2.4GHz band

ExploiTR, 2018-06-26:
**Hmm. I've said the AP is detected normally**

tobozo, 2018-06-26:
Please try to understand what [dual band](https://www.lifewire.com/dual-band-wireless-networking-explained-818279) is, review the settings on your AP, then read the [wiki](https://github.com/spacehuhn/esp8266_deauther/wiki) and you'll see the deauther can only see devices using the 2.4GHz band.

ExploiTR, 2018-06-26:
:sob: :sob: :sob: :sob: I do understand. Look, the AP is a[ Linksys E1200 N300](https://www.linksys.com/gb/p/P-E1200/) Router which only supports that 2.4 GHz band & channels 1-13.
And the stations are, desktop, one laptop and 2 phones which only support that 2.4GHz.
It worked last night. But, not working now :sob:

ExploiTR, 2018-06-26:
The problem is persisting on the ESP side.

- did scan without touching settings : **worked**
- did scan by `stopAP` : **_0 result null X_**
- did scan by `startAP` :  **1 device detected (3 connected)**

ExploiTR, 2018-06-26:
```
Scanning WiFi [15%]:   1 packets/s |  1 devices |  0 deauths
Scanning WiFi [30%]:   2 packets/s |  1 devices |  0 deauths
Scanning WiFi [45%]:   0 packets/s |  1 devices |  0 deauths
Scanning WiFi [60%]:   3 packets/s |  1 devices |  0 deauths
Scanning WiFi [75%]:   1 packets/s |  1 devices |  0 deauths
Scanning WiFi [90%]:   1 packets/s |  1 devices |  0 deauths
Scan results saved in /scan.json
Scan results saved in /scan.json
Removed all APs
Cleared station list
Scan results saved in /scan.json

Scanning WiFi [15%]:   0 packets/s |  0 devices |  0 deauths
Scanning WiFi [30%]:   0 packets/s |  0 devices |  0 deauths
Scanning WiFi [45%]:   0 packets/s |  0 devices |  0 deauths
Scanning WiFi [60%]:   0 packets/s |  0 devices |  0 deauths
Scanning WiFi [75%]:   0 packets/s |  0 devices |  0 deauths
Scanning WiFi [90%]:   0 packets/s |  0 devices |  0 deauths
Scan results saved in /scan.json
Scan results saved in /scan.json
Removed all APs
Cleared station list
```

Look, it detects one now then none.. :sob:

tobozo, 2018-06-26:
>  Looks like spacehunn will tag WTF again. :trollface:

don't worry I cant this for @spacehuhn 

what kind of activity do you generate on the given devices in order to actually have packets to sniff ?

ExploiTR, 2018-06-26:
>what kind of activity do you generate on the given devices in order to actually have packets to sniff ?

Didn't understand

spacehuhn, 2018-06-26:
In other words, are there any packets it could be sniffing? Because maybe there is actually 0 packets per second on the channel the ESP is sniffing on.

ExploiTR, 2018-06-26:
Doesn't it iterate through the channels by default? :fearful: @spacehuhn

ExploiTR, 2018-06-26:
However, my AP is on channel 1, and the stations too..

ExploiTR, 2018-06-26:
@spacehuhn  | I've turned off settings echo `set serialEcho false`

```
Mounting SPIFFS...OK
Switched to Channel 1
Settings loaded from /settings.json
Settings saved in /settings.json
Device names loaded from /names.json
SSIDs loaded from /ssids.json
Scan results saved in /scan.json
Serial interface enabled
Started AP
[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '1', hidden: false, captive-portal: true
STARTED! \o/
v2.0.5
Executing /autostart.txt
Done executing script
Stopped scan
Scan results saved in /scan.json
Removed all APs
Cleared station list
Starting scan for access points (Wi-Fi networks)...
[===== Access Points =====]
ID SSID                             Name             Ch RSSI Enc. Mac               Vendor   Selected
=====================================================================================================
 0 TimeExecutor                                       6  -30 WPA2 ##:##:##:##:##:## BelkinIn         
=====================================================================================================
Stopped scan
Scan results saved in /scan.json
Selected access point TimeExecutor
Stopped scan
Scan results saved in /scan.json
Starting Scan for stations (client devices) - 15s
Stopped Access Point
Scanning WiFi [20%]:   2 packets/s |  0 devices |  0 deauths
Scanning WiFi [40%]:   0 packets/s |  0 devices |  0 deauths
Scanning WiFi [60%]:   0 packets/s |  0 devices |  0 deauths
Scanning WiFi [80%]:   0 packets/s |  0 devices |  0 deauths
[===== Stations =====]
Station list is empty :(
Started AP
Stopped scan
Scan results saved in /scan.json
restarting in 0s - type stop to disable the continuous mode
Stopped scan
Scan results saved in /scan.json
Starting Scan for stations (client devices) - 15s
Stopped Access Point
Scanning WiFi [20%]:   0 packets/s |  0 devices |  0 deauths
Scanning WiFi [40%]:   0 packets/s |  0 devices |  0 deauths
Scanning WiFi [60%]:   0 packets/s |  0 devices |  0 deauths
Scanning WiFi [80%]:   0 packets/s |  0 devices |  0 deauths
Started AP
Stopped scan
Scan results saved in /scan.json
```

spacehuhn, 2018-06-26:
@ExploiTR but why?! Now we can't see what command you typed, which is kinda important for debugging you know... 😉

ExploiTR, 2018-06-26:
@spacehuhn , I've updated logcat

spacehuhn, 2018-06-26:
Thanks. Have you tried running the scanner somewhere else where there are more networks? Have you tried sniffing with other devices to make sure there are packet's beeing sent?

ExploiTR, 2018-06-26:
let me find a way @chicken . In my village, getting a thousand networks are a dream.

ExploiTR, 2018-06-26:
Well, opened virtual interfaces on my WRT.

```
[===== Access Points =====]
ID SSID                             Name             Ch RSSI Enc. Mac               Vendor   Selected
=====================================================================================================
 0 dd-wrt_vap2                                        6  -35    - ##:##:##:##:##:30                  
 1 dd-wrt_vap                                         6  -35    - ##:##:##:##:##:3e                  
 2 TimeExecutor                                       6  -36 WPA2 ##:##:##:##:##:3d BelkinIn         
 3 dd-wrt_vap3                                        6  -36    - ##:##:##:##:##:31                  
=====================================================================================================
Stopped scan
Scan results saved in /scan.json
Starting Scan for stations (client devices) - 15s
Stopped Access Point
Scanning WiFi [20%]:  14 packets/s |  1 devices |  0 deauths
Scanning WiFi [40%]:  19 packets/s |  1 devices |  0 deauths
Scanning WiFi [60%]:  17 packets/s |  1 devices |  0 deauths
Scanning WiFi [80%]:  12 packets/s |  1 devices |  0 deauths
[===== Stations =====]
ID MAC               Ch Name             Vendor   Pkts     AP                               Last Seen Selected
==============================================================================================================
 0 ##:##:##:##:##:75  6                  HaoCheng        8 dd-wrt_vap3                      <1min             
 1 ##:##:##:##:##:35  6                  LiteonTe        2 dd-wrt_vap2                      <1sec             
==============================================================================================================
Started AP
Stopped scan
Scan results saved in /scan.json
```

_And the result is quite normal for the other 2 open networks_ .**The 3rd device connected to my main network which is encrypted by WPA* doesn't seem to be detected**. :fearful: 

But, I'm quite sure that it's software isn't configured to use encrypted frames as, it worked last night :confused: :thinking:

ExploiTR, 2018-06-26:
Also, got this exception : 

```
Exception 2: InstructionFetchError: Processor internal physical address or data error during instruction fetch
Decoding 41 results
0x40229d14: String::~String() at C:\Users\@@@@@\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\cores\esp8266/WString.cpp line 124
0x40214693: Scan::save(bool) at C:\Users\@@@@@\AppData\Local\Temp\arduino_build_862080\sketch/Scan.cpp line 332
0x4022c355: netif_remove at /Users/igrokhotkov/projects/esp8266/esp_iot_sdk_v1.3.0_15_08_08/esp_iot_sdk_v1.3.0/app/lwip/core/netif.c line 249
0x40233fe6: wifi_get_opmode at ?? line ?
0x40232e8e: eagle_lwip_if_free at ?? line ?
0x40234b85: wifi_station_disconnect at ?? line ?
0x40249f5f: wifi_station_stop at ?? line ?
0x4023402d: wifi_get_opmode_default at ?? line ?
0x4023472d: wifi_station_ap_number_set at ?? line ?
0x40234165: wifi_set_broadcast_if at ?? line ?
0x4023484e: wifi_station_set_config_current at ?? line ?
0x402341d2: wifi_set_opmode_current at ?? line ?
0x40221c5b: ESP8266WiFiClass::_mode(WiFiMode) at C:\Users\@@@@@\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\libraries\ESP8266WiFi\src/ESP8266WiFi.cpp line 952
0x40221cf6: ESP8266WiFiClass::disconnect(bool) at C:\Users\@@@@@\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\libraries\ESP8266WiFi\src/ESP8266WiFi.cpp line 952
0x40221d0b: ESP8266WiFiClass::disconnect(bool) at C:\Users\@@@@@\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\libraries\ESP8266WiFi\src/ESP8266WiFi.cpp line 952
0x40229dba: String::reserve(unsigned int) at C:\Users\@@@@@\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\cores\esp8266/WString.cpp line 146
0x40229dea: String::copy(char const*, unsigned int) at C:\Users\@@@@@\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\cores\esp8266/WString.cpp line 175
0x40229348: Print::write(unsigned char const*, unsigned int) at C:\Users\@@@@@\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\cores\esp8266/Print.cpp line 42
0x40226a81: Print::write(char const*) at C:\Users\@@@@@\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\cores\esp8266/Print.h line 60
0x40233fe6: wifi_get_opmode at ?? line ?
0x4023573e: wifi_promiscuous_enable at ?? line ?
0x4022941c: Print::println() at C:\Users\@@@@@\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\cores\esp8266/Print.cpp line 131
0x4021daee: stopAP() at C:\Users\@@@@@\AppData\Local\Temp\arduino_build_862080\sketch/functions.h line 37
0x402148d0: Scan::start(unsigned char, unsigned int, unsigned char, unsigned int, bool, unsigned char) at C:\Users\@@@@@\AppData\Local\Temp\arduino_build_862080\sketch/Scan.cpp line 332
0x40207b59: Accesspoints::print(int, bool, bool) at C:\Users\@@@@@\AppData\Local\Temp\arduino_build_862080\sketch/Accesspoints.cpp line 55
0x401016e3: ets_timer_arm_new at ?? line ?
0x4021496c: Scan::start(unsigned char) at C:\Users\@@@@@\AppData\Local\Temp\arduino_build_862080\sketch/Scan.cpp line 332
0x402016a3: delay at C:\Users\@@@@@\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\cores\esp8266/core_esp8266_wiring.c line 70
0x40214ce8: Scan::update() at C:\Users\@@@@@\AppData\Local\Temp\arduino_build_862080\sketch/Scan.cpp line 332
0x40226c94: esp_yield at C:\Users\@@@@@\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\cores\esp8266/core_esp8266_main.cpp line 43
0x40226d08: __yield at C:\Users\@@@@@\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\cores\esp8266/core_esp8266_main.cpp line 43
0x40106970: free at C:\Users\@@@@@\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\cores\esp8266/libc_replacements.c line 212
0x40226d4c: optimistic_yield at C:\Users\@@@@@\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\cores\esp8266/core_esp8266_main.cpp line 43
0x402262d4: HardwareSerial::available() at C:\Users\@@@@@\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\cores\esp8266/HardwareSerial.cpp line 541
0x4021946a: SerialInterface::update() at C:\Users\@@@@@\AppData\Local\Temp\arduino_build_862080\sketch/SerialInterface.cpp line 121 (discriminator 1)
0x4022536d: DNSServer::processNextRequest() at C:\Users\@@@@@\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\libraries\DNSServer\src/DNSServer.cpp line 51
0x402093c1: DigitalLed::update() at C:\Users\@@@@@\AppData\Local\Temp\arduino_build_862080\sketch/DigitalLed.cpp line 107
0x4021ebac: loop at C:\Users\@@@@@\AppData\Local\Temp\arduino_build_862080\sketch/functions.h line 37
0x40226cb5: esp_schedule at C:\Users\@@@@@\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\cores\esp8266/core_esp8266_main.cpp line 43
0x40226cdd: loop_wrapper at C:\Users\@@@@@\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\cores\esp8266/core_esp8266_main.cpp line 43
0x40100114: cont_norm at C:\Users\@@@@@\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\cores\esp8266/cont.S line 109
```

spacehuhn, 2018-06-26:
Looks like it's working

ExploiTR, 2018-06-26:
?? How? I'm continuously trying it for hours. But, it doesn't even sometimes the other two on an open network isn't showing.

I've already ordered the NodeMCU V3 to test if the problem is persisting in this V2 one. \O/

killergeek, 2018-06-26:
@ExploiTR yea i see the problem. its between the chair and the computer.
and something something RTFM.

jLynx, 2018-06-26:
Closed issue due to the issue being with the user, not the actual software. Sadly we don't provide help for this specific "clearly beta" version of user

closed 2018-06-26 by jLynx

ExploiTR, 2018-06-27:
@killergeek Welcome mate. Looks like went a long way through for the first comment. And, Secondly, be stubborn wouldn't help one well?

And, @jLynx @spacehuhn  **the issue is with the software**. The way it detects stations isn't efficient. I know why this happened, and it's actually a bug. The devices aren't releasing any packet to save power somehow. So, as an open-source and without-warranty project, deauther wouldn't be detecting 100% of the devices using wifi tech and even not of them which **don't** use frame-encryption.

jLynx, 2018-06-27:
I dont see why you reacted to your own comment...
[image]

jLynx, 2018-06-27:
also @killergeek  you just got TOLD, sit the f**k down kid! make a project before commenting 😂

spacehuhn, 2018-06-27:
This might look rude but let me document my reactions reading this word salat

> And, @jLynx @spacehuhn the issue is with the software. 

oookaayy.... 

> The way it detects stations isn't efficient.

rly?!

> I know why this happened, and it's actually a bug.

Well now I'm hyped, tell me what you found young padawan!

> The devices aren't releasing any packet to save power somehow. 

🤦‍♂️ 🤦‍♀️  how is that a bug with the software then?! We already told you that you can only detect devices that send packets while your sniffing for those packets. It's common sense. You can't hear someone that isn't saying anything.

> So, as an open-source and without-warranty project, deauther wouldn't be detecting 100% of the devices using wifi tech and even not of them which don't use frame-encryption.

Oh jeez...

1. `open-source and without-warranty project` has nothing to do with the rest of your comment, so what's your point here?
2. `deauther wouldn't be detecting 100% of the devices using wifi tech` yes it can only detect active devices that use Wi-Fi. (kinda self explanatory, isn't it?)
3. `even not of them which don't use frame-encryption.` Sniffing and detecting devices is absolutly unrelated to the Wi-Fi network encryption beeing used. There is a thing called MAC header in every 802.11 frame.

Some extra notes:
- We talk about a $2 chip here that wasn't meant for this specific purpose, so don't expect great sniffing performance.
- It might not get every packet because of the antenna beeing used
- You lose packets due to channel hopping
- You lose packets because the ESP is doing a lot of other stuff too, not just only sniffing
- The SDK 2.0.0 beeing used might not have the best performance either

tobozo, 2018-06-27:
> make a project before commenting

Actually this thread is the aftermath of doing exactly [that](https://github.com/ExploiTR/DeAutherDroid/issues/1)

Unless stated in the [Code of conduct](https://github.com/spacehuhn/esp8266_deauther/blob/master/CODE_OF_CONDUCT.md), using emojis instead of a GPG public key to sign messages does not invalidate the opening of an issue, I'd be more blaming github for that, why did they allow self-emojis in the first place ?

Anyway, most symptoms described here are from a *different* environment from the usual standard we hear about in this issue tracker.

The build @ExploiTR is trying to achieve is a Java client running on an Android and using the serial to talk to the deauther.
As a result there are mixed problems and mixed symptoms, hence the WTF smell and the fun reactions.

spacehuhn, 2018-06-27:
To sum this whole thing up:
I can't see anything wrong with the serial output you provided. You need devices that send packets and are connected to an AP that is in the list of the ESP8266. The more active the devices are, the higher the chance the ESP will find them.

If it is a software bug, tell us how to fix it.

ExploiTR, 2018-06-27:
By saying `not efficient` I wanted actually to mean it's not official and wouldn't always be successful to detect all the devices.

>If it is a software bug, tell us how to fix it.

I don't know C/C++. And, I'm not an expert software engineer. But, one thing I can suggest. That is,
`airodump-ng` , I mean the whole `aircrack-ng` package is written in **C** . And, I didn't see it failing to detect any stations being active or not. Will this help you?

spacehuhn, 2018-06-27:
Aircrack suit has the same "problems". It can only see active devices. Only advantage is that it lists unconnected devices. But that wouldn't make sense here, since you can only attack connected devices.
Aircrack might give you more reliable results because it'susing dedicated hardware with much much more horsepower than this $2 chip and a real OS to schedule tasks and processes.

ExploiTR, 2018-06-27:
Okay. Thanks