SpacehuhnTech/esp8266_deauther #907: Please add Finnish language to code

developerfromjokela, 2018-06-30:
Hello! I just finished translating `.LANG` file. Can someone include it to the code?

Download .LANG file : https://drive.google.com/open?id=183fpQtOjdQ1Blm_tUHqhd4VnDVnxTyyk

tobozo, 2018-06-30:
hello and thanks for your contribution :+1: 

Can you please post it on a public network (i.e [jsoneditoronline](https://jsoneditoronline.org/) or [pastebin](https://pastebin.com) instead ?

developerfromjokela, 2018-06-30:
There is Pastebin: https://pastebin.com/6sf19Sj3
And JsonEditorOnline: https://jsoneditoronline.org/?id=e7227f11d86b4ddbb63553e7d7669da5

tobozo, 2018-06-30:
thanks!
Since you pasted the JSON twice, I only kept the first block.
I also had to fix some `\n`'s as it would not validate in the editor.

developerfromjokela, 2018-06-30:
So Can you still include it in the code?

tobozo, 2018-06-30:
[it's in the source tree now](https://github.com/spacehuhn/esp8266_deauther/commit/976d31247b909e6548d4c189cdc3f843360998a5), not sure when and if it will land in a release though since there's a [big rewrite](#846) in progress

jimmibond, 2018-07-11:
Here is on json formatter. http://jsonformatter.org/bbca7b

tobozo, 2018-07-11:
this one looks much better than jsoneditoronline.org :-)

Did you change anything since the [commit](https://github.com/spacehuhn/esp8266_deauther/commit/976d31247b909e6548d4c189cdc3f843360998a5)?

developerfromjokela, 2018-08-21:
No, I didn't change anything.

developerfromjokela, 2018-08-21:
If More Text needed to be translated, I can do it.

developerfromjokela, 2019-02-07:
Hello! 
I can't switch to Finnish by using Deauther.me DNS. Also I tried using IP and it worked.

Regards,
@developerfromjokela