SpacehuhnTech/esp8266_deauther #909: Hebrew Translation

itsik-b, 2018-07-03:
I've translated lang file for HEBREW language. I've managed to create the he.lang.gz file and created the webfile.h with the webConverter.py utility.
but it doesn't seem to work. I've set the WEBSPIFFS option in SETTINGS <enabled> and confirmed that the file he.lang.gz copied to SPIFFS.  
* before coping the files to SPIFFS, I got msg "lang he.lang.gz file not found" on serial monitor. after copying it there isn't any message for missing lang file, but still works in english.

What am I missing ?
[he.lang.gz](https://github.com/spacehuhn/esp8266_deauther/files/2158840/he.lang.gz)
[he.lang.txt](https://github.com/spacehuhn/esp8266_deauther/files/2158849/he.lang.txt)

tobozo, 2018-07-03:
Some of the slashed quotes are mistyped (found one with a space between / and "), this makes the JSON parsing fail and the webUI loads the default language.

You can use [this tool](https://jsoneditoronline.org/) to pre-validate your JSON.
Or lazy fix: replace all slashed \" quotes by directional signs such as « and »

itsik-b, 2018-07-03:
I removed all escape signs... still not good.
I've tried to change to RU lang, didn't change either, may be the problem is the lack of support on utf-8 files ?

tobozo, 2018-07-03:
Please submit your changes so we can cross-verify (i.e use gist or json validator online or pastebin).

I doubt there's a problem with utf8 otherwise we'd already know.

If the file is valid JSON and correctly zipped then it's a full browser problem, but I guess you already have Hebrew installed on the phone you're using to test this, right ?

One way to test the ESP is doing its part is to directly access the he.lang(.gz) file from your mobile browser.

Another lazy test: overwrite a language file you know is working (e.g. German) with the hebrew file, then load this language from the webUI, and tell us what you get.

itsik-b, 2018-07-04:
Attached the clean lang file, I can't find errors in it.

Hebrew lang installed on my browsers :) (desktop and mobile phone)

I've tested it from the ESP (http://deauth.me/web/lang/he.lang.gz) the JSON file was loaded O.K. :
[image]

[he_clean.lang.txt](https://github.com/spacehuhn/esp8266_deauther/files/2162095/he_clean.lang.txt)
[he.lang.gz](https://github.com/spacehuhn/esp8266_deauther/files/2162096/he.lang.gz)

itsik-b, 2018-07-04:
Is there a way to make the text right alignment ? 

[image]

tobozo, 2018-07-04:
there are different approaches to that

1) RTL page `html { direction: rtl; }` 
2) RTL layout `section { direction: rtl; }`
3) RTL blocks `span { direction :rtl; }` 

Note that depending on the layout you sometimes have to use both directions in the same document: 

```
 html { direction: ltr; } 
 span { direction: rtl; }
```

Full RTL page will probably trigger the directional anomalies you usually find on LTR-only layouts, where parts are explicitely using text-align:left.

If you come to modify the css, make sure you prefix all your extra directives like this:

```
[lang="he"] div.blah > span.meh { 
  direction: rtl; 
  text-align:left 
}
```

itsik-b, 2018-07-04:
I don't want to mess the other LTR languages.
I can add before every RTL text  **&#x202E;** (Unicode Bidirectional) which will work as RTL text.
But I need to change the alignment of the paragraphs. any ideas how to do it without affecting the other LTR langs ?