SpacehuhnTech/esp8266_deauther #912: Why Can't I see the ADD button....

itsik-b, 2018-07-05:
I'm not able to add a specific SSID to SAVED SSIDs ... missing ADD button.

[image]

spacehuhn, 2018-07-05:
If you open the website with a device with a small screen, you have to open the website in desktopmode to access everything. THe mobile view is simplified.

itsik-b, 2018-07-05:
I see now that on SSID when you add manually it works fine, but when you "Clone Selected APs" it ignores the "Number", and clones it 60 times.

spacehuhn, 2018-07-05:
Clone selected APs is different to adding SSIDs manual. It will fill the list until it's full.
But I see why it can be confusing, I added it on the todo list for the next update :)

closed 2019-10-09 by tobozo