SpacehuhnTech/esp8266_deauther #915: Getting error on every action

yagarea, 2018-07-10:
Hello, I am getting this errors after clicking on most of butttons:

ERROR: loading file: run?cmd=save scan
ERROR: loading file: run?cmd=attack -d
ERROR: loading file: run?cmd=save ssids
ERROR: loading file: attack.json

tobozo, 2018-07-10:
this is mostly normal, please refer to most online tutorials on the topic or read the wiki where most situations are explained

closed 2018-07-10 by tobozo

spacehuhn, 2018-07-10:
TL;DR

The access point and web server hosting the web interface can be unstable depending on the sorrunding WiFi traffic and if you start a scan or attack, you might have to reconnect. If you start an attack with multiple targets on different channels, you won't be able to reconnect until the attack is stopped.