SpacehuhnTech/esp8266_deauther #917: Dstike D-duino V5 +Ring Shiel boot loop

biobizz, 2018-07-13:
Hi,
I'm in a boot loop after flashing.
I try ESP8266_Deauther_v2.0.5_DSTIKE_D-duino_with_LED_Ring_Shield.bin and Compiling using Arduino IDE but the result is alwayus a bootloop
The erroir is this:
"""
Soft WDT reset

ctx: cont 
sp: 3fff1880 end: 3fff1bf0 offset: 01b0

>>>stack>>>
3fff1a30:  3fff1a80 00000001 3fff1c20 40202b1f  
3fff1a40:  3fff3310 00000000 3fff1c20 40202d9f  
3fff1a50:  3fff3310 00000001 3fff1af0 40262bf4  
3fff1a60:  3ffe9274 3fff1af0 00000001 40227154  
3fff1a70:  3ffe9274 00000100 3fff1aa0 40229171  
3fff1a80:  3ffe00ff 3fff1af0 3fff1aa0 40229149  
3fff1a90:  3fff4970 3fff1af0 00000001 4021f10e  
3fff1aa0:  3ffe92f8 00000000 000003e8 4022a032  
3fff1ab0:  3fff49d0 3fff4a20 3fff1ae0 4022a134  
3fff1ac0:  3ffe9274 00000000 3fff1b50 4022a158  
3fff1ad0:  3fff4960 3fff1b40 3fff1b50 40220395  
3fff1ae0:  3fff49b0 0000000f 0000000e 40229faa  
3fff1af0:  00000000 0000204f 0000204f 4022a002  
3fff1b00:  3fff1c14 3fff1b50 3fff1b50 4022a032  
3fff1b10:  3fff1c14 3fff080c 3fff1b50 4022a134  
3fff1b20:  000000b4 3fff1c20 3fff080c 40262bf4  
3fff1b30:  3fff1c14 3fff080c 3fff080c 4021a655  
3fff1b40:  3ffe87d0 402279fc 00000000 00000fa0  
3fff1b50:  3fff4970 0000000f 0000000e 3fff4970  
3fff1b60:  0000000f 0000000e 3fff5d40 0000020f  
3fff1b70:  00000201 00000001 3fff0af8 40229664  
3fff1b80:  3fff0874 3fff0af8 3fff0af8 40229685  
3fff1b90:  3fff1c14 3fff0af8 3fff088c 40262bf4  
3fff1ba0:  3fff1c14 3fff0af8 3fff080c 402207e6  
3fff1bb0:  00000000 00000000 00000000 00000000  
3fff1bc0:  00000000 00000000 00000000 3fff0bbc  
3fff1bd0:  3fffdc20 00000000 3fff0bb5 40226d9e  
3fff1be0:  00000000 00000000 3fff0bd0 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
system param error
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff 00 00 ff ff 00 00 00 00 00 61 10 40 34 27 ff 3f 94 04 00 00 00 00 00 00 
80 1d 20 40 00 01 00 00 34 27 ff 3f 2a a3 22 40 ff ff ff ff ff ff 01 00 ed fe 3f 2b 04 00 00 2b 
04 00 00 0c 02 10 40 8c ed fe 3f d0 84 fe 3f 01 00 00 00 8c 06 10 40 fd 03 00 00 37 a0 22 40 01 
00 00 00 8c 69 10 40 92 9e 22 40 a8 eb fe 3f d0 84 fe 3f 01 00 00 00 00 ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff 00 ff ff ff ff ff ff 0a 00 00 00 45 53 50 5f 42 37 43 41 41 42 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 00 00 04 ff ff ff 01 00 ff ff 
00 00 00 00 00 61 10 40 34 27 ff 3f 94 04 00 00 00 00 00 00 80 1d 20 40 00 01 00 00 34 27 ff 3f 
2a a3 22 40 00 ed fe 3f 2b 04 00 00 2b 04 00 00 0c 02 10 40 8c ed fe 3f d0 84 fe 3f 01 00 00 00 
8c 06 10 40 fd 03 00 00 37 a0 22 40 01 00 00 00 8c 69 10 40 92 9e 22 40 a8 eb fe 3f d0 84 fe 3f 
01 00 00 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 00 00 08 04 0c 11 1c 00 00 00 01 01 
ff ff ff ff 03 00 00 00 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff f""

I have seen the old posts and the cause would seem to be dodgy pin definitions in the config.
I already try to disable all Led and the display but nothing help.

any ideas?
thanks.

darkgrue, 2018-10-17:
What additional information would be helpful? As far as I know, I'm following the recipe in the Wiki.

I'm using:
- Arduino IDE v1.8.5
- esp8266 by ESP8266 Community version 2.0.0
- arduino-esp8266-deauther by tobozo version 2.0.0-deauther

As I mentioned previously, the target board is a [DSTIKE WiFi Packet Monitor V2(Preflashed)](https://www.tindie.com/products/lspoplove/dstike-wifi-packet-monitor-v2preflashed/) on a [DSTIKE Ring Shield](https://www.tindie.com/products/lspoplove/dstike-ring-shield/).

The `config.h` looks like:
```
#ifndef config_h
#define config_h

// ===== DISPLAY LIBRARY ===== //
#include <Wire.h>
#include <SPI.h>
#include "SH1106Wire.h"
#include "SSD1306Wire.h"
#include "SH1106Spi.h"
#include "SSD1306Spi.h"
// =========================== //

// ===================== LED CONFIG ==================== //
/* = = = = = = = = = = = = = = = = = = = = = = = = = = = = = =
    Uncomment the type of LED  you're using
    Only one of them can be defined at a time!
   = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = */
//#define DIGITAL_LED
//#define RGB_LED
#define NEOPIXEL_LED

/* = = = = = = = = = = = = = = = = = = = = = = = = = = = = = =
    Adjust following settings for your type of LED
    you can ignore the rest of the #define's
   = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = */
   
// === Settings for Digital LED and RGB LED === //
#define LED_ANODE false   // common vcc (+)
#define LED_PIN_R 16      // NodeMCU on-board LED
#define LED_PIN_G 255     // 255 = LED disabled
#define LED_PIN_B 2       // ESP-12 LED

// === Settings for RGB LED and Neopixel LED === //
#define LED_MODE_BRIGHTNESS 10 // brightness of LED modes 

// === Settings for Neopixel LED === //
#define LED_NEOPIXEL_NUM 12
#define LED_NEOPIXEL_PIN 15
#define LED_NEOPIXEL_MODE NEO_GRB + NEO_KHZ800
//   NEO_KHZ800  800 KHz bitstream (most NeoPixel products w/WS2812 LEDs)
//   NEO_KHZ400  400 KHz (classic 'v1' (not v2) FLORA pixels, WS2811 drivers)
//   NEO_GRB     Pixels are wired for GRB bitstream (most NeoPixel products)
//   NEO_RGB     Pixels are wired for RGB bitstream (v1 FLORA pixels, not v2)
//   NEO_RGBW    Pixels are wired for RGBW bitstream (NeoPixel RGBW products)

// ===================== DISPLAY CONFIG ==================== //
#define USE_DISPLAY true // default display setting

/* = = = = = = = = = = = = = = = = = = = = = = = = = = = = = =
    Adjust the pins to match your setup
    Comment out the buttons you don't use.
   = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = */
#define BUTTON_UP 12
#define BUTTON_DOWN 13
#define BUTTON_A 14

//#define BUTTON_LEFT 12
//#define BUTTON_RIGHT 13
//#define BUTTON_B 10

/* = = = = = = = = = = = = = = = = = = = = = = = = = = = = = =
    Uncomment the type of display you're using
    Only one of them can be defined at a time!
   = = = = = = = = = = = = = = = = = = = = = = = = = = = = = =
    0x3c = I2C address
    SDA = GPIO 5 (D1)  
    SCL/SCK = GPIO 4 (D2) */
#define DEAUTHER_DISPLAY SSD1306Wire display = SSD1306Wire(0x3c, 5, 4); // for 0.96" OLED
//#define DEAUTHER_DISPLAY SH1106Wire display = SH1106Wire(0x3c, 5, 4); // for 1.3" OLED

/*  RST = GPIO 5 (D1)
    DC = GPIO 4 (D2)
    CS = GPIO 15 (D8) or GND
    SCK/SCL = GPIO 14 (D5)
    SDA/MOSI = GPIO 13 (D7) */
//#define DEAUTHER_DISPLAY SSD1306Spi display = SSD1306Spi(5, 4, 15); // for 0.96" OLED with SPI
//#define DEAUTHER_DISPLAY SH1106Spi display = SH1106Spi(5, 4, 15); // for 1.3" OLED with SPI

//#define FLIP_DIPLAY // uncomment that to flip the display vertically
// ========================================================= //

#endif
```

And in the `Tools` menu, I'm selecting:
```
Board: "DSTIKE by Travis Lin"
Flash Mode: "DIO"
Flash Frequency: "80MHz"
Upload Using: "Serial"
CPU Frequency: "160 MHz"
Flash Size: "1M (256K SPIFFS)"
Reset method: "nodemcu"
Upload Speed: "115200"
```

spacehuhn, 2018-10-17:
Ok thank you.
Can you try to upload the latest version? I pushed some commits the last few days.
It also includes a new config files: https://github.com/spacehuhn/esp8266_deauther/tree/master/configs

If it still boot-loops please post the exception it throws here so maybe I can try to debug it.

darkgrue, 2018-10-20:
Pulled the latest version from GitHub and moved the `configs\DSTIKE_D-duino_with_LED_Ring_Shield\config.h` over to the main sketch.

I had to also remove and re-install the `arduino-esp8266-deauther by tobozo version 2.0.0-deauther` board definition (to remove the internal ArduinoJson library) and installed the `ArduinoJson by Benoit Blanchon Version 5.13.3`.

No joy. LCD is blank, and it's showing it's looping on the serial console:

```
OK

Soft WDT reset

ctx: cont 
sp: 3fff1a40 end: 3fff1ea0 offset: 01b0

>>>stack>>>
3fff1bf0:  40001f46 00000000 00000000 000003c0  
3fff1c00:  30343238 00000300 40203468 3fff39d0  
3fff1c10:  3fff1ed0 3fff3680 000bb200 40205d09  
3fff1c20:  00000005 3fff1c50 00000000 00000000  
3fff1c30:  00000001 3fff4c84 3fff1ed0 3fff35bc  
3fff1c40:  00000002 3fff1cb0 3fff1cb0 3ffe9b29  
3fff1c50:  40002514 3fffdd3c 00000000 00002020  
3fff1c60:  3fff1d30 00000002 00000004 00000000  
3fff1c70:  00000000 00000000 00000000 3fff1c00  
3fff1c80:  00000000 3fff1ed0 3fff35b8 402058bb  
3fff1c90:  3fff1cb0 3fff1cb0 00000004 4010178e  
3fff1ca0:  00002030 3fff33a8 00000000 00000000  
3fff1cb0:  3ffe9b28 00002030 00000000 3fffbff0  
3fff1cc0:  00000000 00000000 3fff4c78 00000000  
3fff1cd0:  3fff35b8 401069de 3fff1da0 00000001  
3fff1ce0:  3fff1d30 00000001 3fff1ed0 4022b032  
3fff1cf0:  00002010 00000001 3fff1da0 4022b08a  
3fff1d00:  3fff35b8 00000001 3fff1da0 4022b455  
3fff1d10:  3ffe9694 3fff1da0 00000001 40267b08  
3fff1d20:  3ffe9694 3fff1da0 00000001 4022b4d4  
3fff1d30:  3ffe00ff 3fff1da0 3fff1d50 4022a1d1  
3fff1d40:  3fff4c18 3fff1da0 00000001 4021e186  
3fff1d50:  3ffe9718 00000000 000003e8 4022b0ba  
3fff1d60:  3fff4c78 3fff4cc8 3fff1d90 4022b1bc  
3fff1d70:  3ffe9694 00000000 3fff1e00 4022b1e0  
3fff1d80:  3fff4c08 3fff1df0 3fff1e00 4021f501  
3fff1d90:  3fff4c58 0000000f 0000000e 4022b032  
3fff1da0:  00000000 0000200f 0000200f 4022b08a  
3fff1db0:  3fff1ec4 3fff1e00 3fff1e00 4022b0ba  
3fff1dc0:  3fff1ec4 3fff0ac4 3fff1e00 4022b1bc  
3fff1dd0:  000000b4 3fff1ed0 3fff0ac4 40267b08  
3fff1de0:  3fff1ec4 3fff0ac4 3fff0ac4 402197a5  
3fff1df0:  3ffe8900 40227008 00000000 00000fa0  
3fff1e00:  3fff4c18 0000000f 0000000e 3fff4c18  
3fff1e10:  0000000f 0000000e 3fff5dd8 000001ff  
3fff1e20:  000001f6 00000001 3fff0db0 4022a6ec  
3fff1e30:  3fff0b2c 3fff0db0 3fff0db0 4022a70d  
3fff1e40:  3fff1ec4 3fff0db0 3fff0b48 40267b08  
3fff1e50:  3fff1ec4 3fff0db0 3fff0ac4 4021f7aa  
3fff1e60:  00000000 00000000 00000000 00000000  
3fff1e70:  00000000 00000000 00000000 3fff0e74  
3fff1e80:  3fffdc20 00000000 3fff0e6d 402263aa  
3fff1e90:  00000000 00000000 3fff0e80 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Mounting SPIFFS...OK
Formatting SPIFFS...OK

Soft WDT reset

ctx: cont 
sp: 3fff1b40 end: 3fff1ea0 offset: 01b0

>>>stack>>>
3fff1cf0:  00000001 00000000 3fff1ed0 40202d9f  
3fff1d00:  3fff35b8 00000001 3fff1da0 4022b455  
3fff1d10:  3ffe9694 3fff1da0 00000001 40267b08  
3fff1d20:  3ffe9694 00000100 3fff1d50 402267e6  
3fff1d30:  3ffe00ff 3fff1da0 3fff1d50 4022a1d1  
3fff1d40:  3fff4c18 3fff1da0 00000001 4021e18b  
3fff1d50:  3ffe9718 00000000 000003e8 4022b0ba  
3fff1d60:  3fff4c78 3fff4cc8 3fff1d90 4022b1bc  
3fff1d70:  3ffe9694 00000000 3fff1e00 4022b1e0  
3fff1d80:  3fff4c08 3fff1df0 3fff1e00 4021f501  
3fff1d90:  3fff4c58 0000000f 0000000e 4022b032  
3fff1da0:  00000000 0000200f 0000200f 4022b08a  
3fff1db0:  3fff1ec4 3fff1e00 3fff1e00 4022b0ba  
3fff1dc0:  3fff1ec4 3fff0ac4 3fff1e00 4022b1bc  
3fff1dd0:  000000b4 3fff1ed0 3fff0ac4 40267b08  
3fff1de0:  3fff1ec4 3fff0ac4 3fff0ac4 402197a5  
3fff1df0:  3ffe8900 40227008 00000000 00000fa0  
3fff1e00:  3fff4c18 0000000f 0000000e 3fff4c18  
3fff1e10:  0000000f 0000000e 3fff5dd8 000001ff  
3fff1e20:  000001f6 00000001 3fff0db0 4022a6ec  
3fff1e30:  3fff0b2c 3fff0db0 3fff0db0 4022a70d  
3fff1e40:  3fff1ec4 3fff0db0 3fff0b48 40267b08  
3fff1e50:  3fff1ec4 3fff0db0 3fff0ac4 4021f7aa  
3fff1e60:  00000000 00000000 00000000 00000000  
3fff1e70:  00000000 00000000 00000000 3fff0e74  
3fff1e80:  3fffdc20 00000000 3fff0e6d 402263aa  
3fff1e90:  00000000 00000000 3fff0e80 40100114  
<<<stack<<<
```
etc., ad nauseam...

I also attempted to upload the precompiled binary `ESP8266_Deauther_v2.0.5_DSTIKE_D-duino_with_LED_Ring_Shield.bin` with the command: `"C:\Users\username\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.6/esptool.exe" -vv -cd nodemcu -cb 115200 -cp COM4 -ca 0x00000 -cf "D:\Users\username\Downloads\ESP8266_Deauther_v2.0.5_DSTIKE_D-duino_with_LED_Ring_Shield.bin"` with the same results.

Let me know if I can provide better info, or try another way to assist with troubleshooting.

spacehuhn, 2018-10-21:
I'm kinda out of ideas...

But maybe you could try the [reset sketch](https://github.com/spacehuhn/esp8266_deauther/tree/master/Reset_Sketch)
and then upload the default config or bin file.

If you have a second ESP8266, it would also help to know if you can flash that one successfully or if the same error occurs.

Or maybe changing some of the [upload settings](https://github.com/spacehuhn/esp8266_deauther/wiki/Installation#upload-settings) fixes it?

darkgrue, 2018-10-21:
> I'm kinda out of ideas...

Yup, me too. I think I've been following the recipes faithfully, but there's some difference between the config for the packet monitor and deauther sketches that is definitely tripping me up.

Photo of the board to clarify the board version:

[image]

> But maybe you could try the reset sketch
and then upload the default config or bin file.

Reset sketch appears to update and run correctly.

Although, with or without doing that first, loading the packet monitor sketch with the following configuration change works every time:

```
//SSD1306 display(0x3c, 5, 4);   // GPIO 5 = D1, GPIO 4 = D2
SH1106 display(0x3c, 5, 4);
```

> If you have a second ESP8266, it would also help to know if you can flash that one successfully or if the same error occurs.

I've got a second specimen of the same board, it behaves the same way.

On the other hand, a [DSTIKE WiFi Deauther MiNi](https://www.tindie.com/products/lspoplove/dstike-wifi-deauther-mini/) does upload and run correctly with this (modified) `config.h`:

```
#ifndef config_h
#define config_h

// ====================== DISPLAY LIBRARY ====================== //
#include "SH1106Wire.h"

// ========================= LED CONFIG ======================== //
#define HIGHLIGHT_LED 16

#define NEOPIXEL_LED

#define LED_MODE_BRIGHTNESS 10 // brightness of LED modes 

#define LED_NEOPIXEL_NUM 1
#define LED_NEOPIXEL_PIN 15
#define LED_NEOPIXEL_MODE NEO_GRB + NEO_KHZ800

// ======================= BUTTON CONFIG ======================= //
#define BUTTON_UP 12
#define BUTTON_DOWN 13
#define BUTTON_A 14

// ======================= DISPLAY CONFIG ====================== //
#define USE_DISPLAY true // default display setting

#define DEAUTHER_DISPLAY SH1106Wire display = SH1106Wire(0x3c, 5, 4); // for 1.3" OLED
// ============================================================= //

#endif
```

With a [DSTIKE Ring Shield](https://www.tindie.com/products/lspoplove/dstike-ring-shield/) and this config (I'm not sure about the numeracy of the `LED_NEOPIXEL_NUM`, I'm assuming the Neopixel on the board counts as 12 + 1) works as well:

```
#ifndef config_h
#define config_h

// ====================== DISPLAY LIBRARY ====================== //
#include "SH1106Wire.h"

// ========================= LED CONFIG ======================== //
#define HIGHLIGHT_LED 16

#define NEOPIXEL_LED

#define LED_MODE_BRIGHTNESS 10 // brightness of LED modes 

#define LED_NEOPIXEL_NUM 13
#define LED_NEOPIXEL_PIN 15 // D8
#define LED_NEOPIXEL_MODE NEO_GRB + NEO_KHZ800
//   NEO_GRB     Pixels are wired for GRB bitstream (most NeoPixel products)
//   NEO_KHZ800  800 KHz bitstream (most NeoPixel products w/WS2812 LEDs)

// ======================= BUTTON CONFIG ======================= //
#define BUTTON_UP 12 // D6
#define BUTTON_DOWN 13 // D7
#define BUTTON_A 14 // D5

// ======================= DISPLAY CONFIG ====================== //
#define USE_DISPLAY true // default display setting

#define DEAUTHER_DISPLAY SH1106Wire display = SH1106Wire(0x3c, 5, 4); // for 1.3" OLED
// ============================================================= //

#endif
```

darkgrue, 2018-10-22:
I've tried playing around with the `config.h` file, even trying the most minimal configuration, digital LED with all channels disabled, no buttons, and the display with it set to disabled (I note that even with `#define USE_DISPLAY false` still requires a `#define DEAUTHER_DISPLAY` line).

I'm thinking at this point the problem lies outside of the `config.h`, and the code is making some assumption about the D-duino-B V5 hardware version I have that's unique to it (and the deauther that isn't present in the packet monitor sketch) that is not setting well with it.

closed 2019-10-09 by tobozo