SpacehuhnTech/esp8266_deauther #918: Can't find the pwned and the light doesn't turn on.

chestnutprog, 2018-07-14:
Outputs from the serial port seem that it works well,but I can't fine the pwned and the light doesn't turn on.
```
Mounting SPIFFS...OK
Formatting SPIFFS...OK
Switched to Channel 1
Settings loaded from /settings.json
Settings saved in /settings.json
Device names loaded from /names.json
SSIDs loaded from /ssids.json
Scan results saved in /scan.json
Serial interface enabled
Started AP
[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '1', hidden: false, captive-portal: true

STARTED! \o/
v2.0.5
Executing /autostart.txt
Done executing script
```
And it is normally heating

chestnutprog, 2018-07-14:
After I ran a scan by serial,it crashed and rebooted,but then i could find the pwned!
```
Started AP
Stopped scan
Scan results saved in /scan.json
# help
ERROR: command "help" not found :(

Exception (9):
epc1=0x401017eb epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000001 depc=0x00000000

ctx: cont 
sp: 3fff1760 end: 3fff1bf0 offset: 01a0

>>>stack>>>
3fff1900:  3fff4fc8 4010695e 3fff1980 00000000  
3fff1910:  40106970 3fff31b8 3fff1a48 40229cc2  
3fff1920:  3ffe922c 00000000 3fff1980 40229c74  
3fff1930:  ffffffff 0000000d 3fff1980 40219071  
3fff1940:  40106970 3fff061c 3fff1980 3fff1970  
3fff1950:  3ffe922c 00000003 3fff31b8 402281cc  
3fff1960:  3ffe922c 00000003 3fff061c 40215814  
3fff1970:  00000001 00000000 00000002 40104023  
3fff1980:  3fff4fd8 0000000f 00000000 3ffe9bc0  
3fff1990:  00000001 40102c43 3ffe9bf0 3ffedf8c  
3fff19a0:  401030a0 40103bca 00000001 00000000  
3fff19b0:  00000000 4010399b 00000000 3ffe9bc0  
3fff19c0:  40102b88 00080000 00000000 7fffffff  
3fff19d0:  0000007f 00040000 00000000 7fffffff  
3fff19e0:  4010524f 00080000 00040000 00000000  
3fff19f0:  00000000 4010510e 00000000 00000000  
3fff1a00:  4000050c 3fffc278 40104f68 3fffc200  
3fff1a10:  00000022 40106718 3fff0af8 40226896  
3fff1a20:  40000f83 40106718 3fff0af8 4010674c  
3fff1a30:  40000ea3 40106718 3fffc250 4000050c  
3fff1a40:  40226c12 00000030 3fff4fb8 0000000f  
3fff1a50:  00000001 00000000 00000000 00000000  
3fff1a60:  3fff33c2 00000725 00000000 ffdfffff  
3fff1a70:  ffffffff 3fffc6fc 00000001 3fff0af8  
3fff1a80:  3fff0af8 0000000a 3fff0bbc 00000030  
3fff1a90:  3fff0af8 0000000a 3fff0bbc 00000030  
3fff1aa0:  3fff0af8 0000000a 3fff0bbc 00000030  
3fff1ab0:  00000003 000002be 00000000 ffdfffff  
3fff1ac0:  3fff2980 4010695e 3fff1b30 00000004  
3fff1ad0:  3fff1b30 3fffdc20 3fff0bbc 40229cc2  
3fff1ae0:  3fff2970 00000004 3fff1b30 40229d1a  
3fff1af0:  000000a0 3fffc6fc 3fff1b30 40229d4a  
3fff1b00:  00000004 00000003 3fff1b30 40229e4c  
3fff1b10:  3ffe0070 3fffc6fc 3fff1b80 0000003b  
3fff1b20:  00000004 00000003 3fff1b80 40218e81  
3fff1b30:  3fff2990 0000000f 00000004 3fff2970  
3fff1b40:  0000000f 00000004 3fff0af8 40226284  
3fff1b50:  3fff061c 3fff061c 3fff0af8 40229817  
3fff1b60:  00000100 3fff3e58 3fff1b80 3fff0bbc  
3fff1b70:  3fffdc20 3fff0af8 3fff061c 4021942d  
3fff1b80:  3fff2950 0000000f 00000004 402252cd  
3fff1b90:  00000000 00000000 3fff085c 402093c1  
3fff1ba0:  3fffdc20 3ffef95c 3fff0bb5 3fff0bbc  
3fff1bb0:  3fffdc20 3ffef95c 3fff0bb5 4021eb02  
3fff1bc0:  40226c15 00000000 00000000 3fff0bbc  
3fff1bd0:  3fffdc20 00000000 3fff0bb5 40226c3d  
3fff1be0:  00000000 00000000 3fff0bd0 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Mounting SPIFFS...OK
Switched to Channel 1
Settings loaded from /settings.json
Settings saved in /settings.json
Device names loaded from /names.json
SSIDs loaded from /ssids.json
Scan results saved in /scan.json
Serial interface enabled
Started AP
[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '1', hidden: false, captive-portal: true
STARTED! \o/
v2.0.5
Executing /autostart.txt
Done executing script
# help
[===== List of commands =====]
help
```

Udara-Sampath, 2018-07-19:
Same problem to me

spacehuhn, 2018-07-19:
It shouldn't crash on scan, there was a bug but it should be solved with that version.
The exception it's throwing is also not very helpful.

You guys can try to flash the latest version, but you have to do that with Arduino. The instructions are in the wiki.
I'm working on a big update but that will probably take a while before it's finished.

Udara-Sampath, 2018-07-20:
what about this
Arduino: 1.8.5 (Windows 10), Board: "NodeMCU (ESP-12 Module), 80 MHz,
Serial, 80MHz, DIO, 115200, 4M (1M SPIFFS)"

Archiving built core (caching) in:
C:\Users\N.Y.R.U.T\AppData\Local\Temp\arduino_cache_839259\core\core_deauther_esp8266_nodemcuv2_CpuFrequency_80,UploadTool_esptool,FlashFreq_80,FlashMode_dio,UploadSpeed_115200,FlashSize_4M1M_add65ab5b881ee98456343952beb7878.a
Sketch uses 584230 bytes (55%) of program storage space. Maximum is 1044464
bytes.
Global variables use 56168 bytes (68%) of dynamic memory, leaving 25752
bytes for local variables. Maximum is 81920 bytes.
error: Failed to open COM3
error: espcomm_open failed
error: espcomm_open failed

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

closed 2019-10-09 by tobozo