SpacehuhnTech/esp8266_deauther #924: ESP8266 is auto resetting | help !

e-3-s, 2018-07-18:
Hi there,

I have flashed the bin file onto the chip, using an usb to serial ftdi module.
The flashing went wel. Then I flashed with arduino ide the ino sketch following the instructions.
Also I enabled in the settings.md file the following.

WEB-INTERFACE
bool webInterface = true;
Enables web interface.

WEB-SPIFFS
bool webSpiffs = true
Enables SPIFFS for all web files.
Can lead to longer loading times but it nice if you need to edit the the web files regularly.

Then I restart the chip by removing the power.
I open up the serial monitor in arduino IDE and I see the following..

Mounting SPIFFS...OK
Formatting SPIFFS...OK

and then restarts.

I am power the chip with the ftdi usb adapter module.

So I don't understand what the problem is....
Does somebody know?

Cediddi, 2019-04-14:
I'm having a very similar issue.

I used esptool 2.6
`esptool.py --port /dev/ttyUSB0 erase_flash`
`esptool.py --port /dev/ttyUSB0 write_flash -fm dio 0 ~/Downloads/ESP8266_Deauther_v2.0.5_1MB.bin`

My device is a ESP-01 module with usb flasher.

Apart from e-3-s I did tried external power supply with and without usb flasher as serial connector and the outcome is same.

```
Mounting SPIFFS...OK
Formatting SPIFFS...OK

Soft WDT reset

ctx: cont 
sp: 3fff1890 end: 3fff1bf0 offset: 01b0

>>>stack>>>
3fff1a40:  00000001 00000000 3fff1c20 40202f17  
3fff1a50:  3fff3378 00000001 3fff1af0 4022a0e5  
3fff1a60:  3ffe922c 3fff1af0 00000001 40262924  
3fff1a70:  3ffe922c 00000100 3fff1aa0 40226fde  
3fff1a80:  3ffe00ff 3fff1af0 3fff1aa0 40228e61  
3fff1a90:  3fff49d8 3fff1af0 00000001 4021f24f  
3fff1aa0:  3ffe92b0 00000000 000003e8 40229d4a  
3fff1ab0:  3fff4a38 3fff4a88 3fff1ae0 40229e4c  
3fff1ac0:  3ffe922c 00000000 3fff1b50 40229e70  
3fff1ad0:  3fff49c8 3fff1b40 3fff1b50 402204d1  
3fff1ae0:  3fff4a18 0000000f 0000000e 40229cc2  
3fff1af0:  00000000 0000201f 0000201f 40229d1a  
3fff1b00:  3ffe8790 3fff1b50 3fff1b50 40229d4a  
3fff1b10:  3ffe8790 3fff07dc 3fff1b50 40229e4c  
3fff1b20:  000000b4 3fff1c20 3fff07dc 40262924  
3fff1b30:  3ffe8790 3fff07dc 3fff07dc 4021a795  
3fff1b40:  3ffe8790 40227800 00000000 00000fa0  
3fff1b50:  3fff49d8 0000000f 0000000e 3fff49d8  
3fff1b60:  0000000f 0000000e 3fff5da8 0000020f  
3fff1b70:  00000202 00000001 3fff0af8 4022937c  
3fff1b80:  3fff0844 3fff0af8 3fff0af8 4022939d  
3fff1b90:  3fff1c14 3fff0af8 3fff085c 40262924  
3fff1ba0:  3fff1c14 3fff0af8 3fff07dc 4022092a  
3fff1bb0:  00000000 00000000 00000000 00000000  
3fff1bc0:  00000000 00000000 00000000 3fff0bbc  
3fff1bd0:  3fffdc20 00000000 3fff0bb5 40226c32  
3fff1be0:  00000000 00000000 3fff0bd0 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,3)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
```

This is happening for all 2.x versions. Last working version is 1.6 for me. For that the output is as expected:

```
Settings:
SSID: pwned
SSID length: 5
SSID hidden: 0
password: deauther
password length: 8
channel: 1
Default MAC AP: 2e:3a:e8:42:04:15
Saved MAC AP: 2e:3a:e8:42:04:15
MAC AP random: 0
Scan hidden APs: 1
deauth reson: 1
attack timeout: 300
attack packet rate: 10
client scan time: 15
use built-in LED: 1
channel hopping: 0
multiple APs: 0
multiple Attacks: 0
mac change interval: 4
1s beacon interval: 0
LED Pin: 2
settings saved
Settings:
SSID: pwned
SSID length: 5
SSID hidden: 0
password: deauther
password length: 8
channel: 1
Default MAC AP: 2e:3a:e8:42:04:15
Saved MAC AP: 2e:3a:e8:42:04:15
MAC AP random: 0
Scan hidden APs: 1
deauth reson: 1
attack timeout: 300
attack packet rate: 10
client scan time: 15
use built-in LED: 1
channel hopping: 0
multiple APs: 0
multiple Attacks: 0
mac change interval: 4
1s beacon interval: 0
LED Pin: 2
Trying to sync the MAC addr with settings
Could not sync because settings are not loaded !
nameList cleared
saving SSIDList...done
Attack LED : OFF

Starting WiFi AP:
SSID     : 'pwned'
Password : 'deauther'
-----------------------------------------------

 generating MACs...done

Starting...
```

tobozo, 2019-10-09:
Closing this issue as inactive, feel free to reopen if necessary.

closed 2019-10-09 by tobozo

Cediddi, 2020-10-21:
I tried the deauther today, after a year and a half. 2.1.0 still gives me the same error but 2.5.0rc2 works without a problem. Just wanted to say thanks to all.