SpacehuhnTech/esp8266_deauther #926: Wemos D1 Mini Pro v2 (Aliexpress) not running

JeroenVBH, 2018-07-21:
> Please search for existing (open and closed) issues first to avoid duplicates.  
Also have a look at the [Wiki](https://github.com/spacehuhn/esp8266_deauther/wiki).  
Please put error messages and code ```inside these 3 quotes/grave accents```
(Delete this text when you've read it)
Can I run this on Wemos D1 Pro Mini v2 16M (128M bit) from Aliexpress?

beykansen, 2018-08-07:
Did you resolve this problem? @JeroenVBH

remy99c, 2018-09-04:
I can't get it to work on wemos d1 mini lite, It does work on a nodeMCU.

JeroenVBH, 2018-09-04:
No the issue is still open

spacehuhn, 2018-09-04:
Thee Wemos D1 mini lite uses an ESP8285 with 1MB embedded flash. The 1MB bin file should work. I tested it on all available (original) Wemos boards.

OtakuNekoP, 2018-10-24:
Hey guys, I resolve the problem today.

Do NOT use the **QIO** SPI MODE, you will not boot up like that.

[image]

Use the **DIO** mode instead, and it works fine on my Wemos D1.

[image]

I hope this tip will help you guys. w

JeroenVBH, 2018-10-24:
I also got IT working now! One problem remains, after a powerloss, the WeMos has lost its settings like WiFi SSID and Password.

closed 2019-10-09 by tobozo