SpacehuhnTech/esp8266_deauther #928: NodeMCU 1.0 (ESP-12E Module) still not available in Arduino IDE.

Tommrodrigues, 2018-07-21:
Hey there,

I realise that this issue was originally raised [here](https://github.com/spacehuhn/esp8266_deauther/issues/880) a few months ago; however, it is my opinion that a reasonable solution was never properly found. 

To summarise:
I currently have a NodeMCU 1.0 which I would usually flash in the Arduino IDE using the `NodeMCU 1.0 (ESP-12E Module)` option under the ESP8266 Modules section. To confirm, I have successfully added both `ESP8266 Modules` and `ESP8266 Deauther Modules`. However, I wish to flash the source code provided [here](https://github.com/spacehuhn/esp8266_deauther/releases) onto my NodeMCU as stated in the [Wiki](url
https://github.com/spacehuhn/esp8266_deauther/wiki/Installation#compiling-using-arduino-ide). The problem is that while the wiki shows the option `NodeMCU 1.0 (ESP-12E Module)` as you can see from the picture below located on the Wiki:

[image]

However, unlike as shown on the wiki, the option for the `NodeMCU 1.0 (ESP-12E Module)` is not present under the "ESP8266 Deauther Modules" and I simply have the option of `NodeMCU (ESP-12 Module)` as you can see from my screenshot below:

[image]

On the original issue which I linked above, "tobozo" suggested the following:
> You already added it, just select the ESP12 board and follow the steps in the wiki.

Implying that selecting  the `NodeMCU (ESP-12 Module)` option, this would work for flashing onto an NodeMCU 1.0. However, this option is of course not meant for the NodeMCU 1.0 as it's recommended flash settings differ in a number of areas including CPU Frequency, Flash size and more.

I am not sure if this is a very rare problem as but I was unable to find any more mention of the issue other than that original Issue that I linked above.

My question is: is there any way to make the `NodeMCU 1.0 (ESP-12E Module)` option appear under the "ESP8266 Deauther Modules" as it is supposed to? If not, which settings should I use for flashing the code onto the NodeMCU (i.e. would it work just using the standard `NodeMCU 1.0 (ESP-12E Module)` under the "ESP8266 Modules" or does it have to specifically be under "ESP8266 Deauther Modules" or could I use the `NodeMCU (ESP-12 Module)` as suggested in the original issue; in which case, which settings would need to be adjusted.

Thank you in advance for any help,

Kind regards, 
Tom

P.S. I understand that I am able to flash reliably to my NodeMCU using the "Esptool" tool; however, I believe that this issue should not be forgotten or at least properly addressed in case people do indeed come across it in the future and are unsure as to what the correct solution is.

tobozo, 2018-07-22:
good point about the documentation :+1: 

If you feel like experimenting, the boards listed in your menu depend on the contents of the [boards.txt](https://github.com/tobozo/Arduino/blob/deauther/boards.txt) file located in your `packages/deauther/hardware/esp8266/2.0.0-deauther/` folder.

Otherwise just use the generic_esp8266 profile and play with the values until the results are stable.

happy hacking :hammer: :wrench:

Tommrodrigues, 2018-07-22:
Thanks for your response!

Just to confirm; I see you have attached a link to a deauther SDK screenshot, do you mean that indeed, by selecting the `NodeMCU (ESP-12 Module)` from the "ESP8266 Deauther Modules", this should work for all NodeMCUs despite flash settings etc. differing from that of the normal NodeMCU 1.0 (ESP-12E Module)?  If not, I'll try the `generic_esp8266` as you suggested.

Either way, I'll have a play around with different board selections and settings etc. and get back ASAP.

Thank you once again!

vladagurets, 2018-10-15:
@Tommrodrigues Hi Tomm
I also have ESP-12E module and i can't understand how i can run deauther on it
when i try to upload sketch to module i always get next errors:

> warning: espcomm_send_command: didn't receive command response
> warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
> warning: espcomm_send_command: wrong direction/command: 0x01 0x03, expected 0x01 0x04
> An error occurred while uploading the sketch

Can u help me please? How i can use ESP-12E with this app?

vladagurets, 2018-10-15:
I also played with upload settings (CPU, flash size etc.) but still get error
P.s. also I can’t burn firmware from your release list (get similar error )

tobozo, 2018-10-16:
The SDK has been updated to display separately presets form Nodemcu v1.0 and v0.9.

Since it's not a version increment, your IDE won't get any notification about that update so you may want to manually remove/install the deauther SDK from your boards manager.

vladagurets, 2018-10-16:
The same error

**Params that i use for uploading (ESP8266 Modules)**
All work fine

[image]

**Params that i use for uploading (ESP8266 Deauther Modules)**

[image]

> .warning: espcomm_send_command: didn't receive command response
> warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
> warning: espcomm_send_command: wrong direction/command: 0x01 0x03, expected 0x01 0x04
> An error occurred while uploading the sketch

vladagurets, 2018-10-16:
And i tried a next firmware versions:
1. ESP8266_Deauther_v2.0.5_1MB.bin
2. ESP8266_Deauther_v2.0.5_4MB.bin
3. ESP8266_Deauther_v2.0.5_DSTIKE_Deauther_v3_._DSTIKE_NodeMCU-07_v2.bin
4. esp8266_deauther_1mb.bin

but got the same error:

> Uploading Test Firmware...
> esptool v0.4.8 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
> 	setting board to ck
> 	setting baudrate from 115200 to 9600
> 	setting port from /dev/tty.usbserial to /dev/cu.usbserial-1410
> 	espcomm_upload_file
> 	espcomm_upload_mem
> opening port /dev/cu.usbserial-1410 at 9600
> 	tcgetattr
> 	tcsetattr
> 	serial open
> opening bootloader
> resetting board
> trying to connect
> 	setting character timeout 0
> 	done
> 	setting character timeout 1
> 	done
> 	espcomm_send_command: sending command header
> 	espcomm_send_command: sending command payload
> trying to connect
> 	setting character timeout 0
> 	done
> 	setting character timeout 1
> 	done
> 	espcomm_send_command: sending command header
> 	espcomm_send_command: sending command payload
> 	espcomm_send_command: receiving 2 bytes of data
> 	espcomm_send_command: receiving 2 bytes of data
> 	espcomm_send_command: receiving 2 bytes of data
> 	espcomm_send_command: receiving 2 bytes of data
> 	espcomm_send_command: receiving 2 bytes of data
> 	espcomm_send_command: receiving 2 bytes of data
> 	espcomm_send_command: receiving 2 bytes of data
> 	espcomm_send_command: receiving 2 bytes of data
> Uploading 569696 bytes from to flash at 0x00000000
> 	erasing flash
> 	size: 08b160 address: 000000
> 	first_sector_index: 0
> 	total_sector_count: 140
> 	head_sector_count: 16
> 	adjusted_sector_count: 124
> 	erase_size: 07c000
> 	espcomm_send_command: sending command header
> 	espcomm_send_command: sending command payload
> 	setting timeout 15000
> 	setting character timeout 150
> 	done
> 	setting timeout 1
> 	setting character timeout 1
> 	done
> 	espcomm_send_command: receiving 2 bytes of data
> 	writing flash
> ......warning: espcomm_send_command: didn't receive command response
> warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
> warning: espcomm_send_command: wrong direction/command: 0x01 0x03, expected 0x01 0x04
> closing bootloader
> error: espcomm_upload_mem failed
> 
> Finished with exit code 255

vladagurets, 2018-10-16:
Same issue on the 115200 )

tobozo, 2018-10-16:
maybe check your cable ? change USB port ? make sure you've pushed the flash button on the module ?

Also try to flash a regular, non-deauther sketch, just to make sure the error you get is related to this project, because currently it seems to be an environment problem. and you'll probably find more efficient help on the [relevant gitter](https://gitter.im/esp8266/Arduino)

vladagurets, 2018-10-16:
Maybe i'm dumb
but i can't figure out with uploading app to my module
I have some questions:
1. Does firmware from your releases is required?
2. If it's required how i can choose correct version?
3. Can i burn it with esptool-gui (with testing before uploading)?
4. Flash button should be pressed when firmware is uploading?

[image]
[image]

Sorry for child's questions...

vladagurets, 2018-10-16:
> maybe check your cable ? change USB port ? make sure you've pushed the flash button on the module ?
>
> Also try to flash a regular, non-deauther sketch, just to make sure the error you get is related [...]

Reqular work's fine
if i chose Board ESP-12E from ESP8266 Modules (not deauther's submenu items)

tobozo, 2018-10-16:
The error messages you pasted are from the command line, now you're throwing symptoms from the Arduino IDE.

This is getting confusing, but I'm sure you'll find a way to flash your module :-)

>  Does firmware from your releases is required?

only if none of the other methods worked

>    If it's required how i can choose correct version?

I have no idea, I only use the compiled version

>    Can i burn it with esptool-gui (with testing before uploading)?

sure, tell us if it worked :-)

>    Flash button should be pressed when firmware is uploading?

actually it's in this order:

1) press RST (keep it pressed)
2) press FLASH (keep it pressed)
3) release RST
4) wait 2 seconds
5) release FLASH

When doing this, your ESP is supposed to wait for being flashed.
Usually Arduino IDE will send a signal with the same effect.
So when this fails you can still do it manually.

vladagurets, 2018-10-16:
Thnx a lot!
i ll continue struggling)

vladagurets, 2018-10-17:
It's works now (under Windows)
previously i use my Mac for burning and uploading sketch
i don't know the reason why it's works shitty under Mac
____
thnx for support)

tobozo, 2018-10-17:
that's probably because the [per-OS speeds are different in the boards.txt](https://github.com/tobozo/Arduino/blob/deauther/boards.txt#L155)

vladagurets, 2018-10-17:
oh ok
thanx

closed 2019-02-17 by Tommrodrigues