SpacehuhnTech/esp8266_deauther #938: Hab alles gemacht klappt aber trotzdem nicht ! (Esp8266)

barxara, 2018-08-22:
Ich hab mir die bin Files heruntergeladen und hab als erstes die 1mb bin File mit dem flasher auf das NodeMCU  gemacht mit 80 hz   und der rest standart .. als es fertig war hab ich ihn einmal raus und wieder rein gesteckt .. dasNetze würd auc angezeigt pwned .. hab mich connected bin danin den browser gegangen hab alles gemacht .. wollte dann apis und das ganze machen .. hab alles nach Anleitung gemacht aber das lädt die de extrem langsam ... als ich das gemacht hab stand oben error=cmd... ... hab alles mögliche ausprobiert ... auch tumb bin File drauf gemacht oder mit 40 hz ... klappt nicht .. Und wenn ich dann Station. .. suche also drücke disconnected das wifi und mein normales WLAN connected wieder .. hab es auch schon entfernt damit es nicht connected .... klappt nicht !! Bitte hilft mir

davedarko, 2018-08-23:
Du hast also erfolgreich den Controller programmiert, mit 80MHz und 40MHz und konntest in beiden Fällen das generierte Netzwerk mit der SSID "pwned" sehen und dich auch verbinden. Bis dahin sehe ich noch keinen Grund zu schreiben "alles gemacht, klappt aber trotzdem nicht". Danach wird es mir leider zu kryptisch was deine Denkpunkte angeht, aber ich denke das Verhalten ist normal - das Wiki sagt ja dass es die Verbindung beim Scannen wegen Channelhopping verlieren MUSS. https://github.com/spacehuhn/esp8266_deauther/wiki/Web

barxara, 2018-09-01:
Es funktioniert aber irgendwie überhaupt nicht .. Ich lade die Website und drück auf Lord apis/stations Reload .. also nachdem es blau geladen hat .. = eroor-cmd.. und ich werde com wifi disconnected. . Und mit meinem verbunden.. es hat einmal funktioniert wo mir die wlan stations angezeigt wurden weiter bin ich noch ni e gekommen ... bitte hilft mir !!

spacehuhn, 2018-10-16:
https://github.com/spacehuhn/esp8266_deauther/wiki/FAQ
😃

closed 2018-10-16 by spacehuhn