SpacehuhnTech/esp8266_deauther #945: Bug in SimpleList<T>::binSearch

xdobry, 2018-09-07:
After quite long time I finally found the bug that crashing the app

please add the gard to 
int SimpleList<T>::binSearch
other wise the lastNodeGot could be NULL

      if (hNode) {
        lastNodeGot = hNode;
        lastIndexGot = hIndex;
        isCached = true;
      } 

How to reproduce:
- add one name
- restart application
- make scan aps 
- crash!

Anyway SimpleList.h is not very sophisticated code. Why not reuse some others code.
Generally Using & references would increase the speed of application. Because the structures are not copied.

Example:

T &get(int index);

template<typename T>
T &SimpleList<T>::get(int index){
  Node<T> *hNode = getNode(index);
  return hNode->data;
}

Will return the reference. You do not need to make replace if you update the node
As in

void Names::internal_deselect(int num) {
    Device newDevice = list->get(num);
    newDevice.selected = false;
    list->replace(num, newDevice);
}

but just

void Names::internal_deselect(int num) {
    list->get(num).selected = false;
}

because you get reference not value.
Problem what to do if hNode is NULL.
 T() in old code is not good idea. I it not work with reference any more.
because the T() is lokal object and will be removed.

template<typename T>
T SimpleList<T>::get(int index) {
    Node<T>* hNode = getNode(index);
    return hNode ? hNode->data : T();
}

In many many cases following as parameter definition.
const String &t
not
String t

The second one will create the copy of t (it means new memory allocation of char *prt and copy it).

bradanlane, 2018-09-12:
Thanks for the debugging. 

When I looked at your fix I noticed several other places which could cause problems. There are several instances of calling getNode() where the result is not tested before the return object is used.

spacehuhn, 2018-09-12:
In my defense, using standard libraries can cause problems in the old SDK and the Arduino LinkedList library has similar bugs as well.

I'm working on version 3 that has a much better codebase than the current deauther. 
If you like to make a PR to fix the described bug, I'll merge it. Otherwise please be patient :) 

Big *THANKS* for your debugging work!

bradanlane, 2018-09-12:
I’ll be patient :-)

I am not fully confident with the breadth of my changes.

spacehuhn, 2018-10-16:
So I wanted to refactor basically everything and came pretty far but the old SDK 2.0.0 with its compiler is just not very efficient. More on it here: https://github.com/spacehuhn/esp8266_deauther/issues/846

I updated the SimpleList, let me know if you find another bug :D

tobozo, 2019-10-09:
Please update to the latest version of SimpleList library and reopen this issue if necessary.

closed 2019-10-09 by tobozo