SpacehuhnTech/esp8266_deauther #947: espcomm error

Acamar71, 2018-09-12:
I tryed to use Flash Download Tools and Arduino in both i get ```warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_open failed```
and i tryed all the fixes on here (holding flash button, trying different COM ports
 and installing drivers)

tobozo, 2018-09-12:
you're giving up too fast and don't provide enough information to get significant help here, also it does not seem to be a deauther problem but more of a cable / com port / ESP problem.

if you keep on trying (e.g. google your error message) you'll eventually find out what's causing this, and then hold on to the next error, and so on.

happy hacking!

spacehuhn, 2018-10-16:
https://www.youtube.com/watch?v=53zkBvL4ZB4

closed 2018-10-16 by spacehuhn