SpacehuhnTech/esp8266_deauther #948: ERROR: timeout loading file run?cmd=scan aps -ch 1

michaelnj1, 2018-09-12:
Hi, I bought a preflashed board straight from DSTRIKE on AliExpress and upon getting the board I am greeted with multiple errors when trying to scan for APs. I connect to the network and agree to the terms just fine, however upon loading I get the error ```ERROR: loading file: run?cmd=save names``` and once scanning for APs I get ```ERROR: timeout loading file run?cmd=scan aps -ch 1```. I thought buying preflashed would avoid these problems but I guess not. I'm using the WiFi Deauther v3.5. Any and all help is appreciated.

sk3tch4d, 2018-10-06:
Same issue here. WiFi Deauther USB. Getting the following errors:

ERROR: timeout loading file run?cmd=scan aps -ch all 
ERROR: timeout loading file run?cmd=scan stations -t 15s -ch all
and 
ERROR: timeout loading file run?cmd=save names (for a second before the other error)

Same errors with change of variables such as channel and seconds. Not working

spacehuhn, 2018-10-16:
Without further information, I can just link you to the [FAQ](https://github.com/spacehuhn/esp8266_deauther/wiki/FAQ).

closed 2019-10-09 by tobozo