SpacehuhnTech/esp8266_deauther #950: DSTike d-duino V5

alexlab963, 2018-09-15:
Hey, I just received my d-duino v5. Upon powering it up I didn't see any new wifi networks so I flashed it in Arduino IDE with the latest 2.0.5 release. I use DSTIKE for as my board and have tried both the default and the OLED config but I still do not see pwned in my wifi list. There is, however, an unsecured ESP network that I am unable to join. Any help?

NapsterMD, 2018-09-25:
Download .bin file v1.6 with 1 mb.. name of file esp8266_deauther_1mb.bin.. size (556 kb)..
Use NodeMCU-flasher with these settings:
Baudrate: 115200
Flash size: 1MByte
Flash speed: 80MHz
SPI Mode: DIO
in the CONFIG bar of NodeMCU-flasher chose the file and address must begin at 0x00000 and than press FLASH..

spacehuhn, 2018-10-16:
Were you able to get it running?

The installation docs are here:
https://github.com/spacehuhn/esp8266_deauther/wiki/Installation
https://github.com/spacehuhn/esp8266_deauther/wiki/Setup-Display-&-Buttons

Kaio-Dev, 2019-06-30:
Don't want to necro a post. But is there support for DSTIKE D-duino-B V5, with the Deauth v2.1.0?
Only thread I could find which is similar to what im looking for.

tobozo, 2019-10-09:
Closing this issue as probably not applicable due to lack of memory, feel free to reopen if necessary.

closed 2019-10-09 by tobozo