SpacehuhnTech/esp8266_deauther #952: wifi_send_pkt_freedom() return -1

liuyiner3, 2018-09-20:
I have some questions to ask .I compiled and flash on my esp8266 using 2.0.6 source code, and the scanning and other features were fine. But，When I use deauth to attack my ap，it doesn‘t work。I am sure it’s 2.4g。
  And I use monitor got this information：
# attack -d
Start attacking
[Pkt/s] All:    0 | Deauths:   0/20  | Beacons:   0/0   | Probes:   0/0  
[Pkt/s] All:    0 | Deauths:   0/20  | Beacons:   0/0   | Probes:   0/0  
[Pkt/s] All:    0 | Deauths:   0/20  | Beacons:   0/0   | Probes:   0/0  
[Pkt/s] All:    0 | Deauths:   0/20  | Beacons:   0/0   | Probes:   0/0  
[Pkt/s] All:    0 | Deauths:   0/20  | Beacons:   0/0   | Probes:   0/0  
[Pkt/s] All:    0 | Deauths:   0/20  | Beacons:   0/0   | Probes:   0/0  
[Pkt/s] All:    0 | Deauths:   0/20  | Beacons:   0/0   | Probes:   0/0  

And I debug the program find that wifi_send_pkt_freedom() return -1。But I don‘t know how to fix this problem。
Can you give me some help。
Thinks a lot！！！！！！！！

spacehuhn, 2018-09-21:
Weird.
Have you compiled it with the correct board selected?
https://github.com/spacehuhn/esp8266_deauther/wiki/Installation#compiling-using-arduino-ide

liuyiner3, 2018-09-21:
> Weird.
> Have you compiled it with the correct board selected?
> https://github.com/spacehuhn/esp8266_deauther/wiki/Installation#compiling-using-arduino-ide

I have already fix my boread。I compiled use a wrong A_config.h ，so that deauth doesn't work.

thanks a lot！！！

closed 2018-09-21 by spacehuhn