SpacehuhnTech/esp8266_deauther #960: ESP8266 shows network "AI-Thinker" instead of "pwned"

P2Koyz, 2018-10-10:
Hello, so I have recently purchased the esp8266, yet when I power it on and search for wifi networks, pwned does not come up and instead the esp8266 projects the network "AI-Thinker". Do I have the wrong nodeMCU? Also, when I connect to "AI-Thinker", I cannot connect to 192.168.4.1 or 192.168.4.2. Any help would be great. I have also tried following these tutorials:
https://www.instructables.com/id/DIY-Wifi-Jammer-With-ESP8266-and-Mobile-App/
https://www.teachmemicro.com/nodemcu-wifi-jammer/
And each of these just leads to a string of errors and missing libraries. On youtube videos documenting connecting to "pwned", they dont have to go through any arduino process at all, and it works as intended as soon as its powered on which leads me to believe I have the wrong NodeMCU.

spacehuhn, 2018-10-10:
Don't follow these shitty tutorials, but have a look at the official wiki:

https://github.com/spacehuhn/esp8266_deauther/wiki/Installation

That should explain you everything.

P2Koyz, 2018-10-11:
> Don't follow these shitty tutorials, but have a look at the official wiki:
> 
> https://github.com/spacehuhn/esp8266_deauther/wiki/Installation
> 
> That should explain you everything.

Ive spent the last 6 hours trying to get this fucker to work, only God knows all the random Chinese software Ive installed, random useless tutorials and "fixes" that just lead to bug after bug, and it STILL didnt work. It took me a solid 7 minutes reading your article and following the tutorial to get this to work. You are doing Gods work

spacehuhn, 2018-10-11:
Haha glad to hear that it worked 😄

Delfino520, 2018-10-15:
> Don't follow these shitty tutorials, but have a look at the official wiki:
> 
> https://github.com/spacehuhn/esp8266_deauther/wiki/Installation
> 
> That should explain you everything.

@spacehuhn Hi, bro I want to introduce a interesting project to you. I just get the code online in China, it is a firmware working on ESP8226, the work principle is similar with linset, wifiphisher or fusion, the different is it works on ESP8226, i call it WIFI BAIT. I am not developer and i do not know anything about programming. There are no similar projects on Github. I think it is a interest project with unknown author, but i confirm that the author is a Chinese people, i have translated Chinese construction into English, if you are interested in it, please contract me. I hope you can put it on Github and try to improve it with developer from all over the world. If you decide to improve this project, i strongly recommend you to combine the webpages function with this project, because in this project you cannot choose webpage you like, such as Browser Connection Reset, Firmware Upgrade page, Browser plugin update and so on. By the way the webpage in this project is too ugly.I have attach all the document & software needed in the email, hope can hear you soon. I have sent all the project code and your email address, if you did not received, please contact me, my email address: s386106832@gmail.com

closed 2018-10-15 by spacehuhn