SpacehuhnTech/esp8266_deauther #962: v2.1.0 - Station mac's not scrolling left/right

Daedelos, 2018-10-16:
When displaying Stations the mac is not scrolled left & right in order to display the last hex digit that is off the right hand edge of the oled as in the prior release.

tobozo, 2018-10-16:
Scrolling isn't part of the specs so it's not like it's missing, and another approach to "fix" this could be to use a larger display.

You can still use the serial console or the web console to get the full MAC string anyway :-)

Daedelos, 2018-10-16:
The previous release (2.0.5/2.0.6) scrolled/moved the Station ssid & mac left right in order to display the full mac
e.g. Microsof:a1:b2:c3:d4 the 4 falls off the end of a 128x64 oled

In order to handle that the previous version moved Microsof:a1:b2:c3:d4 left then right to display the nibble off the screen,

The current version does not have this action. Has this action been removed?

tobozo, 2018-10-16:
oops I hadn't noticed that, maybe @spacehuhn can make a statement about the disappearance of this feature ?

spacehuhn, 2018-10-16:
It's not a feature, it's a big (never thought I would say it this way around). Will look into it later.

spacehuhn, 2018-10-16:
Pushed something, please confirm if the scrolling works for you now 😃

Daedelos, 2018-10-16:
Looks good - circular scrolling working well. Thanks again.

closed 2018-10-16 by Daedelos