SpacehuhnTech/esp8266_deauther #964: Cannot run program "{runtime.tools.esptool.path}/esptool": error=2, No such file or directory

samofta, 2018-10-17:
I've been trying to upload the Arduino IDE code to the Lolin NodeMCU ESP8266, but have failed.

I get this in the log:

```
Arduino: 1.8.7 (Mac OS X), Board: "NodeMCU 1.0 (ESP-12E Module), 160 MHz, Serial, 80MHz, DIO, 9600, 1M (256K SPIFFS)"

In file included from sketch/CLI.h:18:0,
                 from /Users/samofta/Downloads/esp8266_deauther-master 2/esp8266_deauther/esp8266_deauther.ino:29:
sketch/DisplayUI.h:84:42: warning: large integer implicitly truncated to unsigned type [-Woverflow]
         const uint8_t scrollSpeed      = 500; // time interval in ms

                                          ^
Sketch uses 703124 bytes (92%) of program storage space. Maximum is 761840 bytes.
Global variables use 42402 bytes (51%) of dynamic memory, leaving 39518 bytes for local variables. Maximum is 81920 bytes.
java.io.IOException: Cannot run program "{runtime.tools.esptool.path}/esptool": error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at java.lang.Runtime.exec(Runtime.java:620)
	at java.lang.Runtime.exec(Runtime.java:485)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:11)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:207)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:82)
	at processing.app.SketchController.upload(SketchController.java:736)
	at processing.app.SketchController.exportApplet(SketchController.java:703)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2028)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:247)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 10 more
An error occurred while uploading the sketch

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

What am I doing wrong? I followed all instructions on the Installation page but simply cannot figure out how to do this. I have also installed the drivers for the CH340G.

FaiyazPatwari, 2018-10-17:
Caused by: java.io.IOException: error=2, No such file or directory --> Addressing Issue Sir,

samofta, 2018-10-17:
What does that mean? The memory addresses are incorrect?

samofta, 2018-10-18:
Oh well, just realized how stupid I am. I had flashed the firmware to the ESP8266, and then tried to upload the code again from the Arduino IDE. Closing.

closed 2018-10-18 by samofta