SpacehuhnTech/esp8266_deauther #966: user_interface.h

Raellbv, 2018-10-19:
??  user_interface.h ??
i tray to uploud but missing the the file
===========================================================
Arduino:1.8.7 (Windows Store 1.8.15.0) (Windows 10), Board:"Arduino Yún"

esp8266_deauther:9:30: error: user_interface.h: No such file or directory

compilation terminated.

exit status 1
user_interface.h: No such file or directory
============================================================

spacehuhn, 2018-10-19:
# You have to select the right board under Tools -> Board as it is explained in the [wiki](https://github.com/spacehuhn/esp8266_deauther/wiki/Installation#compiling-using-arduino-ide)

Namya125, 2019-05-14:
In file included from C:\Users\Namya Priya\Documents\Arduino\libraries\uMQTTBroker/uMQTTBroker.h:4:0,

from C:\Users\Namya Priya\Documents\Arduino\uMQTTBrokerSampleOOFull\uMQTTBrokerSampleOOFull.ino:9:

C:\Users\Namya Priya\Documents\Arduino\libraries\uMQTTBroker/user_interface.h:37:2: error: #error LWIP_OPEN_SRC must be defined

 #error LWIP_OPEN_SRC must be defined

  ^
I am using esp32 as an mqtt  broker for which I am getting an error as above
how do I rectify this error.

tobozo, 2019-05-14:
@Namya125 replace your esp32 by an esp8266 and the error should be fixed

Namya125, 2019-05-14:
> @Namya125 replace your esp32 by an esp8266 and the error should be fixed

is there any other method to rectify this error?

tobozo, 2019-05-14:
@Namya125 yes you can eventually use an [esp32 as a bridge to an esp8266](https://github.com/tobozo/M5Stack-ESP12-Shield)

Namya125, 2019-05-15:
@tobozo thank you!! will try it.

closed 2019-10-09 by tobozo